KejPi/AbracaDABra #129: Logging of DLS (and maybe DL+)

andimik, 2024-04-16:
We can save slides from SLS, or log the complete SPI, but when recording, a separate txt-file with the DLS (and/or DL+) texts with a time stamp will improve long recording.

This feature was introduced by DAB Player under Windows some years ago.

KejPi, 2024-04-16:
Do you have any proposal for the format of the file?

andimik, 2024-04-16:
I don't care.

```
Date Time {tab} DL {tab} artist {tab} title
```

or similar...

KejPi, 2024-04-17:
Just a side note - when you do audio recording in AAC or MP2 format, DL(+) is inside the file in so called X-PAD. You can use for example [XPADxpert](https://www.basicmaster.de/xpadxpert/) application to check it. In most of the cases slides are also inside because they are rarely broadcasted in dedicated data service. So at the end it would be probably much more interesting to develop player that would play audio and display associated DL(+) and slides ;-)

andimik, 2024-04-17:
I know, that's what I am doing. But it will fail loading when the file is too big for xpadxpert.

KejPi, 2024-04-17:
I see, will implement the exporting when I have some time, but now I have to finalize TII, but it will still take a lot of time :-(

closed 2024-09-13 by KejPi