KejPi/AbracaDABra #130: Build will fail at transmitter map

andimik, 2024-05-03:
I understood that the map is not working for Ubuntu 22.04 (qt < 6.5.0)

```
Qt version < 6.5.0, transmitter map will not be available in TII dialog
```
but I cannot build the devel branch at all:

```
andreas@E535:~/apps/AbracaDABra/build$ cmake ..
-- Version: v2.5.0-44-gd89fe48
-- Build type not specified: defaulting to release.
-- libdabsdr found: /home/andreas/apps/AbracaDABra/lib/linux_x86_64/cmake/dabsdr.cmake
-- Checking for one of the modules 'rtlsdr'
-- Checking for one of the modules 'usb-1.0'
-- Checking for one of the modules 'faad'
-- Checking for one of the modules 'mpg123'
-- Checking for one of the modules 'airspy'
-- Checking for one of the modules 'qcustomplot'
CMake Warning at CMakeLists.txt:505 (message):
  Qt version < 6.5.0, transmitter map will not be available in TII dialog

-- Configuring done
-- Generating done
-- Build files have been written to: /home/andreas/apps/AbracaDABra/build
andreas@E535:~/apps/AbracaDABra/build$ make
[  2%] Built target AbracaDABra_lrelease
[ 15%] Built target AbracaDABra_tooling
[ 16%] Automatic MOC and UIC for target AbracaDABra
[ 17%] Built target AbracaDABra_autogen
[ 18%] Running AUTOMOC file extraction for target AbracaDABra
[ 18%] Built target AbracaDABra_automoc_json_extraction
[ 19%] Running moc --collect-json for target AbracaDABra
[ 20%] Automatic QML type registration for target AbracaDABra
[ 21%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGButton_qml.cpp
[ 22%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGColors_qml.cpp
[ 23%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGHorizontalShadow_qml.cpp
[ 25%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGItem_qml.cpp
[ 26%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGScrollableText_qml.cpp
[ 27%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGSwitch_qml.cpp
[ 28%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGTabBar_qml.cpp
[ 28%] Generating .rcc/qmlcache/AbracaDABra_qml/EPGTabButton_qml.cpp
[ 29%] Generating .rcc/qmlcache/AbracaDABra_qml/TransmitterMarker_qml.cpp
[ 30%] Generating .rcc/qmlcache/AbracaDABra_qml/epg_qml.cpp
[ 31%] Generating .rcc/qmlcache/AbracaDABra_qml/map_blank_qml.cpp
[ 31%] Generating .rcc/qmlcache/AbracaDABra_qml/map_qml.cpp
Error compiling qml file: /home/andreas/apps/AbracaDABra/gui/qml/map.qml:69:26: error: Expected property type
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:278: gui/.rcc/qmlcache/AbracaDABra_qml/map_qml.cpp] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:140: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

KejPi, 2024-05-04:
I will check that in the evening but the QCustomPlot in Ubuntu 22.04 is not compatible either - currently I do not know how to check it's version from cmake. You would need to compile it from source and instruct cmake to take that version.

andimik, 2024-05-04:
I've commented lines 63 - 263 (this is the map) in `map.qml`, it will continue till 96%.

But still no luck.

```
[ 96%] Linking CXX executable AbracaDABra
/usr/bin/ld: CMakeFiles/AbracaDABra.dir/tii/tiidialog.cpp.o: in function `TIIDialog::reset()':
tiidialog.cpp:(.text+0x6033): undefined reference to `QCPGraph::setData(QList<double> const&, QList<double> const&, bool)'
/usr/bin/ld: tiidialog.cpp:(.text+0x60ed): undefined reference to `QCPGraph::setData(QList<double> const&, QList<double> const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:1610: gui/AbracaDABra] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:141: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

KejPi, 2024-05-04:
Yes, this is QCustomPlot issue I mentioned in my previous answer

andimik, 2024-05-04:
I tried these CMake options (thanks to `ccmake`) 

```
 AIRSPY                           OFF
 AIRSPY_INCLUDE_DIRS              /usr/include
 AIRSPY_LINK_LIBRARIES            /usr/lib/x86_64-linux-gnu/libairspy.so
 APPLE_APP_BUNDLE                 OFF
 APPLE_BUILD_X86_64               OFF
 CMAKE_BUILD_TYPE
 CMAKE_INSTALL_PREFIX             /usr/local
 LIBFAAD2_INCLUDE_DIRS            /usr/include
 LIBFAAD2_LINK_LIBRARIES          /usr/lib/x86_64-linux-gnu/libfaad.so
 LIBMPG123_INCLUDE_DIRS           /usr/include
 LIBMPG123_LINK_LIBRARIES         /usr/lib/x86_64-linux-gnu/libmpg123.so
 LIBUSB_LINK_LIBRARIES            /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 PORTAUDIO_LINK_LIBRARIES         /usr/lib/x86_64-linux-gnu/libportaudio.so
** QCUSTOMPLOT                      OFF**
 QCUSTOMPLOT_INCLUDE_DIRS         /usr/include
 QCUSTOMPLOT_LINK_LIBRARIES       /usr/lib/x86_64-linux-gnu/libqcustomplot.so
 QT_ADDITIONAL_HOST_PACKAGES_PR   
 QT_ADDITIONAL_PACKAGES_PREFIX_   
 QT_DIR                           /usr/lib/x86_64-linux-gnu/cmake/Qt6
 Qt6CoreTools_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6CoreTools
 Qt6Core_DIR                      /usr/lib/x86_64-linux-gnu/cmake/Qt6Core
 Qt6DBusTools_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6DBusTools
 Qt6DBus_DIR                      /usr/lib/x86_64-linux-gnu/cmake/Qt6DBus
 Qt6GuiTools_DIR                  /usr/lib/x86_64-linux-gnu/cmake/Qt6GuiTools
 Qt6Gui_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Gui
 Qt6LinguistTools_DIR             /usr/lib/x86_64-linux-gnu/cmake/Qt6LinguistTools
 Qt6Multimedia_DIR                /usr/lib/x86_64-linux-gnu/cmake/Qt6Multimedia
 Qt6Network_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt6Network
 Qt6OpenGL_DIR                    /usr/lib/x86_64-linux-gnu/cmake/Qt6OpenGL
 Qt6Positioning_DIR               /usr/lib/x86_64-linux-gnu/cmake/Qt6Positioning
 Qt6QmlCompilerPlus_DIR           Qt6QmlCompilerPlus_DIR-NOTFOUND          
 Qt6QmlModels_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlModels
 Qt6QmlPrivate_DIR                /usr/lib/x86_64-linux-gnu/cmake/Qt6Qml
 Qt6QmlTools_DIR                  /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlTools
 Qt6QmlWorkerScript_DIR           /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlWorkerScript
 Qt6Qml_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Qml
 Qt6QuickControls2_DIR            /usr/lib/x86_64-linux-gnu/cmake/Qt6QuickControls2
 Qt6QuickTemplates2Private_DIR    /usr/lib/x86_64-linux-gnu/cmake/Qt6QuickTemplates2
 Qt6Quick_DIR                     /usr/lib/x86_64-linux-gnu/cmake/Qt6Quick
 Qt6Svg_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Svg
 Qt6WidgetsTools_DIR              /usr/lib/x86_64-linux-gnu/cmake/Qt6WidgetsTools
 Qt6Widgets_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt6Widgets
 Qt6Xml_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Xml
 Qt6_DIR                          /usr/lib/x86_64-linux-gnu/cmake/Qt6
 RTL_SDR_INCLUDE_DIRS             /usr/local/include
 RTL_SDR_LINK_LIBRARIES           /usr/local/lib/librtlsdr.so
 SOAPYSDR                         OFF
 USE_FDKAAC                       OFF
 USE_PORTAUDIO                    ON
 USE_SYSTEM_AIRSPY                ON
 USE_SYSTEM_FAAD                  ON
 USE_SYSTEM_FDKAAC                ON
 USE_SYSTEM_LIBUSB                ON
 USE_SYSTEM_MPG123                ON
 USE_SYSTEM_PORTAUDIO             ON
** USE_SYSTEM_QCUSTOMPLOT           OFF**
 USE_SYSTEM_RTLSDR                ON
 USE_SYSTEM_SOAPYSDR              OFF
 XKB_INCLUDE_DIR                  /usr/include
 XKB_LIBRARY                      /usr/lib/x86_64-linux-gnu/libxkbcommon.so
```

Now I can at least build it.

KejPi, 2024-05-04:
Yes, but you do not have TII so it is pointless to play with development branch

andimik, 2024-05-04:
Bad news: Qt 6.5 is too new even for Ubuntu 24.04 (which was released some days ago) and will be implemented in October release 24.10 only.

```
$ rmadison qt6-base
 qt6-base | 6.2.4+dfsg-2ubuntu1   | jammy/universe             | source
 qt6-base | 6.2.4+dfsg-2ubuntu1.1 | jammy-updates/universe     | source
 qt6-base | 6.4.2+dfsg-18         | mantic/universe            | source
 qt6-base | 6.4.2+dfsg-21.1build5 | noble/universe             | source
 qt6-base | 6.4.2+dfsg-21.1build5 | oracular/universe          | source
 qt6-base | 6.6.2+dfsg-6~1        | oracular-proposed/universe | source
```

Legend:

Ubuntu 24.04 (Noble Numbat)
Ubuntu 22.04.4 LTS (Jammy Jellyfish)

So we need to wait for AppImages.

KejPi, 2024-05-04:
Yes, I was disappointed with their decision too and wasn't the only one. Qt6.5 was released more than year ago and it is LTS version as Ubuntu 24.04. To be honest I do not understand why they did not include and it was stupid decision IMO. If you want to stay with Ubuntu, you have 3 options - wait for AppImage, install Qt from official source (https://doc.qt.io/qt-6/qt-online-installation.html) (this is what I do) or find some unofficial Ubuntu repo with Qt 6.5.

andimik, 2024-05-04:
I found another - let's say - attempt on https://forum.qt.io/topic/148418/how-to-install-qt6-latest-for-ubuntu-22-jammy-fish/7

see
https://github.com/miurahr/aqtinstall

```
pip install -U pip
pip install aqtinstall
aqt install-qt linux desktop 6.6.0 gcc_64
aqt install-qt linux desktop 6.6.0 --archives qtbase qtmultimedia qtdeclarative qtsvg
```
The challenge is to install qt**6**multimedia. I could not find that :-(

It takes a while, then you need to declare the correct path in cmake:

```
cmake .. -DCMAKE_PREFIX_PATH=/home/andreas/6.6.0
```

The interesting thing is, it warns, that Qt is too old, although 6.6.0 is in use (I think so).

```
-- Found WrapVulkanHeaders: /usr/include  
CMake Warning at CMakeLists.txt:505 (message):
  Qt version < 6.5.0, transmitter map will not be available in TII dialog
```

```
[ 31%] Generating .rcc/qmlcache/AbracaDABra_qml/map_qml.cpp
Error compiling qml file: /tmp/AbracaDABra/gui/qml/map.qml:69:26: error: Expected property type
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:278: gui/.rcc/qmlcache/AbracaDABra_qml/map_qml.cpp] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:140: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2
```

But due to missing qt6multimedia, it does not build.

On https://ddalcino.github.io/aqt-list-server/ you can see all packages.

And then I installed Qt 6.7.0 via online installer, but qcustomplot is not found ...

```
-- Checking for one of the modules 'qcustomplot'
CMake Error at CMakeLists.txt:499 (find_package):
  Could not find a package configuration file provided by "QT" with any of
  the following names:

    Qt6Config.cmake
    qt6-config.cmake

  Add the installation prefix of "QT" to CMAKE_PREFIX_PATH or set "QT_DIR" to
  a directory containing one of the above files.  If "QT" provides a separate
  development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/tmp/AbracaDABra/build/CMakeFiles/CMakeOutput.log".
```

Then I checked https://www.qcustomplot.com/index.php/download, where the latest supported version is 6.4 ...

So I have no idea what to do and might to give up ...

KejPi, 2024-05-04:
aqt can be the way to go, but I would really recommend to download official qt6 online installer, it installs to your home directory and it is very easy to install with simple wizard. Then you only need to take qmake from this installation and compile the app with that. You only need to call cmake with parameter like 

-DCMAKE_PREFIX_PATH=$QT_PATH/lib/cmake

Where QT_PATH is path to your Qt installation/version/compiler 

QCustomPlot can be compiled with QT 6.7 without problems but you need to compile shared library. When I am at my PC I can provide step-by-step instructions for Ubuntu 22.04 if you are interested.

KejPi, 2024-05-06:
devel branch can be built under Ubuntu 22.04 with obsolete Qt version. If you set USE_SYSTEM_QCUSTOMPLOT=OFF when running CMake it should clone QCustomPlot library repo and build it during application compilation. 

WARNING: Make sure that this library is then first in load path or better uninstall qcustomplot library from Ubuntu repository to avoid conflicts. 

NOTE: Map will not be available for Qt < 6.5, there is not way to overcome this limitation because Qt Location was added in Qt 6.5.

KejPi, 2024-05-06:
To build with manually installed Qt, use following CMake command:

`cmake .. -DUSE_SYSTEM_QCUSTOMPLOT=OFF -DCMAKE_PREFIX_PATH=$HOME/Qt/6.5.3/gcc_64/lib/cmake`

Considering version 6.5.3 installed in `$HOME/Qt` (replace with you path to `gcc_64/lib/cmake`)

Then after build I would recommend to run it with X.org instead of default Wayland:

`QT_QPA_PLATFORM=xcb ./gui/AbracaDABra`

Let me know if you need some help.

andimik, 2024-05-07:
Ok, I will (later).

I tried to build the main branch (9bc0bfdc96cc4f207a11b613431511d695d0321e) against the Qt6.7, but when I start the app, the language is English, and not system. Very strange, even when I delete the ini-file.

> QT_QPA_PLATFORM=xcb 

This only works when I install this:

```
sudo apt-get install libxcb-cursor-dev
```

KejPi, 2024-05-07:
It seems it is not correctly reported from the system. I am not sure if I will be able to reproduce it and fix. I hope that at least manual language settings works.

andimik, 2024-05-08:
I tried the latest commit, but I found out that Qt6Positioning was not installed with 6.7.0, but with a different version in the directory `/Qt/Tools/QtDesignStudio/qt6_design_studio_reduced_version/lib/cmake`

So I could successfully compile it with your hint:

```
cmake .. -DUSE_SYSTEM_QCUSTOMPLOT=OFF -DCMAKE_PREFIX_PATH=~/Qt/Tools/QtDesignStudio/qt6_design_studio_reduced_version/lib/cmake
```

[image]

[image]
(German is working now)

Therefore, closing, many thanks for your efforts!

KejPi, 2024-05-09:
I can see already firs issue with Ubuntu again 🙁 It seems that Unicode symbols are not supported by default font - there should be arrow symbol on table header, not rectangle

andimik, 2024-05-09:
Looks good now. Thanks again.

[image]

In 
https://github.com/KejPi/AbracaDABra/blob/1f9105464986615318c7cae73a9ff1ef751b7573/gui/tii/txdataloader.cpp#L43

I fear you mean 2, not 3.

But why not use the same name as in qirx? I needed to rename that file or modify

https://github.com/KejPi/AbracaDABra/blob/1f9105464986615318c7cae73a9ff1ef751b7573/gui/tii/txdataloader.cpp#L105

KejPi, 2024-05-09:
Yes, channel column is typo, thanks for pointing out, but channel column is not used so it does not harm.

I would like to implement automatic download of the database from FMLIST once I get an approval for that. Also I do not have any permission to use data from any other app and I do not actually want to use that data so I have chosen CSV file name for AbracaDABra as it is. 
Until I get an approval from the owner to use data from FMLIST database, it will be left to the responsibility of the user to get the data and to provide them to AbracaDABra in the expected location and format.

andimik, 2024-05-09:
Understood.

Another screenshot, showing it works as expected (with live signals and xml-files)

[image]

BTW: Cool feature, that the channel selector uses the frequency information from the xml-header.

```
<Frequency Unit="kHz" Value="181936"/>
```

KejPi, 2024-05-09:
Thanks, I would like to add also filename parsing to guess channel/frequency in case that XML is not inside.

andimik, 2024-05-09:
When I chose `system`, the map stays in Prague, but the distance to all (!) transmitter is 0 km

[image]

[image]

So I used my GPS mouse on `/dev/ttyACM0` which works (double checked with `gpsmon` tool) as well as the manual setting in my case.

I'm glad you have added a check if somebody enters the coordinates as `xxx,xxx` and not `xxx.xxx` ;-)

KejPi, 2024-05-09:
This is actually an intended behavior. When you choose System and system does not provide valid location, the map is centered in Prague but there is no blue location point. Because application does not have valid location, the distance cannot be calculated so it is set to zero. But I think it would be less confusing for the user to leave it empty or N/A in the table in such case.

Thanks for confirmation that the serial GPS device works :-) I do not have any HW for testing it.

KejPi, 2024-05-09:
Some more hints if you are interested: TII plot can be resized vertically or hidden by dragging the splitter between map and plot. You can zoom in the plot by mouse wheel (both axes) then drag the contents by mouse, or you can zoom only selected axis by clicking on it. Selection is cleared by clicking anywhere in plot area and zoom can be reset on right mouse click.

andimik, 2024-05-09:
Yes, empty would be better instead of zero.

Scrolling in TII works.

But resizing the TII graph below the map seems to consume too much CPU power and will react very very slowly (I even got a warning as AbracaDABra is not responding any more, but clicked on "wait").

KejPi, 2024-05-09:
I do not see any problem on Mac, I will add it to my list of Ubuntu issues :-(

andimik, 2024-05-09:
But maybe this is related to internet connection?

KejPi, 2024-05-09:
TII graph does not use any internet connection only map does. And it has also some cache so the bandwidth usage should not be high. I will try to reproduce the problem when I start my Ubuntu VM.

andimik, 2024-05-09:
BTW: I confirm, that special characters are displayed correctly

[image]

KejPi, 2024-05-09:
I know, we have many of them in czech ;-)

andimik, 2024-05-09:
Any plans to reduce the sensitivity (in order to reduce fakes) or add a slider like in Qirx?

KejPi, 2024-05-09:
As I wrote, I need to do algo tuning and also CPU usage optimizations when it performance optimizations be done. I prefer to avoid any manual controls of the algo.

KejPi, 2024-05-09:
> Yes, empty would be better instead of zero.
> 
> Scrolling in TII works.
> 
> But resizing the TII graph below the map seems to consume too much CPU power and will react very very slowly (I even got a warning as AbracaDABra is not responding any more, but clicked on "wait").

Just be sure - the issue is when you try to move the splitter between map and plot, correct (increasing or decreasing vertical size of graph)? I can se quite slow behavior on my Ubuntu VM (not that app is not responding) but to be honest I am not sure if I can solve it. It seems to be problem of QML/Map because even when I build the app without QCustomPlot support and I try to resize TII Dialog with map only, it is slow too.

andimik, 2024-05-09:
Yes, this slider

andimik, 2024-05-10:
There is another two problems (maybe in Ubuntu only???)

The TII window is in the foreground, even with `Alt + Tab` I cannot switch to the main window. Don't know if you aware of this.

[image]

Remark: The TII window can be placed to the left or right using `Win-Key + Left/Right Arrow`, but I cannot do that with the main window.

[Bildschirmaufzeichnung vom 10.05.2024, 11:12:12.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/4f450ce0-9351-4502-a919-52915c221621)

[Bildschirmaufzeichnung vom 10.05.2024, 11:31:12.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/ec9c9ab5-2c07-461a-869b-9690c3b2f4a3)

KejPi, 2024-05-10:
There is a strange behavior of dialogs on Windows too (domehow related to Qt version)\
Do other dialogs do the same - Settings and Ensemble information?

andimik, 2024-05-10:
I don't know why, but I can shift the main window now.

[Bildschirmaufzeichnung vom 10.05.2024, 12:02:45.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/32dfce8b-6dcf-4d0a-a0dd-be95575d431c)

All four windows,

- Settings
- EPG
- application log
- ensemble information

work, but they all are in the foreground although I can click on the main window and move it.

[Bildschirmaufzeichnung vom 10.05.2024, 11:56:52.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/c60500d5-baae-4aaf-baf3-4cb691d41353)

KejPi, 2024-05-16:
Dialog windows behavior should be fixed now, you can try if you want.

KejPi, 2024-05-31:
@andimik commented in #139:

> Well, I thought I had success with your hints, but this is the output:
> 
> ```
> cmake .. -DUSE_SYSTEM_QCUSTOMPLOT=OFF -DCMAKE_PREFIX_PATH=~/Qt/6.7.0/gcc_64/lib/cmake/
> ```
> 
> ```
> -- Could NOT find Qt6Positioning (missing: Qt6Positioning_DIR)
> CMake Error at CMakeLists.txt:512 (find_package):
>   Found package configuration file:
> 
>     /home/andreas/Qt/6.7.0/gcc_64/lib/cmake/Qt6/Qt6Config.cmake
> 
>   but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
>   FOUND.  Reason given by package:
> 
>   Failed to find required Qt component "Positioning".
> 
>   Expected Config file at
>   "/home/andreas/Qt/6.7.0/gcc_64/lib/cmake/Qt6Positioning/Qt6PositioningConfig.cmake"
>   does NOT exist
> 
>   
> 
>   Configuring with -DCMAKE_FIND_DEBUG_MODE=TRUE might reveal details why the
>   package was not found.
> 
>   Configuring with -DQT_DEBUG_FIND_PACKAGE=ON will print the values of some
>   of the path variables that find_package uses to try and find the package.
> ```

Sorry, I do not want to mix the topics, co let's continue here. 
It is strange because I saw the TII dialog working in your screenshots. 

Well, but the error suggests that you did not include Qt Positioning in your Qt 6.7 installation. You can start Maintenance Tool from Qt installation directory choose Add or remove components and check Qt 6.7 -> Additional libraries -> Qt Positioning.

andimik, 2024-05-31:
I found that in my Qt version this is not existing 

[image]

Let me try to install it from another version.

KejPi, 2024-05-31:
Well, this is happening form time to time. Just uninstall (uncheck) Qt 6.7.0 and install Qt 6.7.1 or 6.5.3 that is LTS version. Just make sure that you add all necessary libraries. This is what I typically install:
[image]

You do not need all for sure, but Multimedia, Positioning, Location, Image Formats, Connectivity, Websockets is the minimum I would say.

KejPi, 2024-06-06:
So what is your status? Are you able to build? I have pushed new version of the library and I would appreciate if you can test it and provide feedback.

andimik, 2024-06-08:
Yes, I could build thanks to your hints. The older Qt version 6.5.3 is working.

[image]

andimik, 2024-08-21:
https://github.com/KejPi/AbracaDABra/blob/095516133078daff5cd33df3b8ca6b3c979eef9d/README.md?plain=1#L236

In order to build `devel` branch, `qt6-positioning-dev` is necessary. So please add it to the `README.md` in the `devel`-branch.

```
sudo apt-get install qt6-positioning-dev
```

Tested under Ubuntu 24.04 (Linux Mint). **No need to install separate Qt in this case** (my Linux partition has crashed and therefore I've re-installed a fresh one), this means it will save a lot of disc space.

It will be `qt6-positioning-dev:amd64 (6.4.2-3build3)` then, and it works as expected.

[image]

KejPi, 2024-08-23:
You can build with Qt6.4 but with the functionality is limited. From my point of view, it is obsolete Qt version.

andimik, 2024-08-23:
`qml6-module-qtcore` is required as well, btw. This is not mentioned in the readme.

closed 2024-09-13 by KejPi