KejPi/AbracaDABra #134: German umlauts are displayed incorrectly.

M4rc-Andre, 2024-05-10:
If the song title contains German umlauts, it is not displayed correctly. See picture.

Wrong: jetzt on air: Die Toten Hosen - WÃ¿nsch dir was
Correct: jetzt on air: Die Toten Hosen - Wünsch dir was

[image]

M4rc-Andre, 2024-05-11:
However, the umlauts are correct on another channel. Could it be that the radio station is encoding the umlauts incorrectly?

[image]

andimik, 2024-05-11:
Yes, please inform the radio station. As I am from Austria, you can believe me that all characters are correctly showed by AbracaDABra.

So please close this issue.

KejPi, 2024-05-12:
I also believe that special characters are handled correctly, we have many special characters in Czech. But if you share a recording, I can inspect it to be sure.

closed 2024-06-13 by KejPi