KejPi/AbracaDABra #135: `Input device error: Device disconnected` when trying to do a band scan

bemxio, 2024-05-21:
Hello! I wanna try and check out AbracaDABra, however upon doing a band scan, it returns an error `Input device error: Device disconnected`, despite my SDR being inside, not touched or moved.

I have tried other DAB+ SDR software and it seems to be cooperating fine with my SDR, so I assume there's a problem somewhere inside AbracaDABra.

My SDR is a generic RTL2832U, and I am running AbracaDABra on Arch Linux from the AUR package.

andimik, 2024-05-21:
When you activate the expert settings and scan on an existing channel, does it work then?

bemxio, 2024-05-21:
Nope, it also gets disconnected when choosing the channel manually.

KejPi, 2024-05-21:
I am sorry you have problems with the app.

Please check usbfs buffer size, although it is probably set correctly if other apps are running, but it is typical issue I have on Arch: https://github.com/KejPi/AbracaDABra?tab=readme-ov-file#usbfs-buffer-size

If this does not help, please run the app from terminal and share it's output.

As last trial you might try to run rtl_tcp and connect through it to see if it works.

bemxio, 2024-05-21:
Ah, I'm so sorry for missing that in the README, after setting the buffer to unlimited, everything's working well. Thank you for pointing it out to me!

closed 2024-05-21 by bemxio

old-dab, 2024-06-02:
The USB buffers are too big. 15 * 1.6 MB is too much for Linux.
I reduced INPUT_CHUNK_IQ_SAMPLES in gui/input/inputdevice.h to (1024 * INPUT_CHUNK_MS). No more problems since then!

KejPi, 2024-06-02:
Thank you for this hint. My intention was to keep low CPU usage by having long buffers and thus less input events, etc. And also avoiding a situation that input buffers are coming faster than application is able to serve. 

Is your modification recommended for all OS or should I use it for Linux only?

old-dab, 2024-06-02:
It works for Windows, too. Qt-DAB and Dabstar for example uses 32 kB buffers, 50 x smaller.

bemxio, 2024-06-02:
> My intention was to keep low CPU usage by having long buffers and thus less input events, etc.

The buffer size could be customizable then, from the settings in expert mode! That would let those with low-end CPUs to still keep the usage low while keeping the plug-and-play experience.

KejPi, 2024-06-02:
This could be an option. I will do some trials to see the impact of smaller buffer size, maybe it will be negligible