KejPi/AbracaDABra #139: Ensemble Information 2nd column shows previous value

andimik, 2024-05-31:
In case you 

a) listen to a station (here: `PRVI` on 10D = `SLO DAB+ R1`)
b) change the DAB channel via dropdown menu (here I switched to 12C = `SLO DAB+ R2W`)
c) but **not** select a service
d) go to the Ensemble Information window

then you see values in the 2nd column (service, service ID, CUs, etc.) from step a) and in the main window a) is highlighted, but not selected.

[image]

I suggest to display `N/A` in such a case.

KejPi, 2024-05-31:
Strange, it does not happen to me using development version on macOS. When you do the steps you describe, service information is cleared. 

Did you build from devel branch or is 2.5.0?

[image]

andimik, 2024-05-31:
Yes, the right column should be cleared/reset when open the window. I need to reset the values as they ALWAYS show errors, even with local signals.

Let me test with a more recent version, but I remember this is the same behavior for several versions ...

KejPi, 2024-05-31:
Yes, I did not touch this functionality for a long time. My screenshot was from development branch I built yesterday and some problem is there too.

I think reseting the stats on opening of the dialog is not desired behavior. With good quality signals there should be only few FIC errors, if it is not the case, there might be some other issue. I will check that too.

andimik, 2024-05-31:
I checked which version I'm using: it's 1229b8a18fb7afd9f6fe91885f9e0977506c816e

Unfortunately (due to the Qplot issues I had in the past), I can no longer build AbracaDABra. Will reinstall my laptop with Ubuntu 24.04 soon.

andimik, 2024-05-31:
I've downloaded the 2.5.0 release AppImage, same behavior.

KejPi, 2024-05-31:
> I checked which version I'm using: it's [1229b8a](https://github.com/KejPi/AbracaDABra/commit/1229b8a18fb7afd9f6fe91885f9e0977506c816e)
> 
> Unfortunately (due to the Qplot issues I had in the past), I can no longer build AbracaDABra. Will reinstall my laptop with Ubuntu 24.04 soon.

I thought the build issue was already resolved in #130 and you were able to build it and run. Please describe your current problems in reply to #130

andimik, 2024-05-31:
Well, I thought I had success with your hints, but this is the output:

```
cmake .. -DUSE_SYSTEM_QCUSTOMPLOT=OFF -DCMAKE_PREFIX_PATH=~/Qt/6.7.0/gcc_64/lib/cmake/
```

```
-- Could NOT find Qt6Positioning (missing: Qt6Positioning_DIR)
CMake Error at CMakeLists.txt:512 (find_package):
  Found package configuration file:

    /home/andreas/Qt/6.7.0/gcc_64/lib/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find required Qt component "Positioning".

  Expected Config file at
  "/home/andreas/Qt/6.7.0/gcc_64/lib/cmake/Qt6Positioning/Qt6PositioningConfig.cmake"
  does NOT exist

  

  Configuring with -DCMAKE_FIND_DEBUG_MODE=TRUE might reveal details why the
  package was not found.

  Configuring with -DQT_DEBUG_FIND_PACKAGE=ON will print the values of some
  of the path variables that find_package uses to try and find the package.
```

KejPi, 2024-06-15:
I cannot reproduce the issue with service label, etc. in second column. I have tried on macOS and on Arch Linux and it does not happen.

andimik, 2024-06-15:
I will rebuild latest commit and let you know.

andimik, 2024-06-16:
Issue is solved now :-)

Thanks!

[image]

andimik, 2024-09-19:
The bug is still alive in 2.5.92 appImage

[image]

KejPi, 2024-09-20:
Well, actually it was never fixed because I could not reproduce it. But I can reproduce it now :-) The key is that the channel you manually tune to without service selection must contain valid ensemble. I think I know where is the problem, it will be fixed in next release.

closed 2024-09-20 by KejPi