KejPi/AbracaDABra #141: Request to add Raw file support for .wav file format?

HeisensOppings, 2024-06-04:
Request to add Raw file support for .wav file format?
Version: 2.5.0

Hi, I use raw files for learning and testing, because I can't receive DAB signals in my area. welle-io supports .wav. The raw file I downloaded from the Internet is in WAV format, which has a DSCTy:60 sub-channel, but Welle-io does not have the announcement epg display function, so I came here

Settings... -> Device -> Raw file -> Open File... only (*.bin *.s16 *.u8 *.raw *.sdr *.uff), but not *.wav

welle-io
https://github.com/AlbrechtL/welle.io

The dab raw file source I can find:
https://www.sdrangel.org/iq-files/
  (Digital Audio Broadcasting (DAB) BBC Radio Multiplex)
https://github.com/williamyang98/DAB-Radio/releases
  (baseband_9A_0.zip
  baseband_9B_0.zip
  baseband_9C_0.zip
  baseband_9C_3.zip
  baseband_dab_uk_sdrangel.zip (BBC))

KejPi, 2024-06-04:
You are right, the file filter is limiting in this case. 
As workaround you can rename the file to *.raw, choose number of bits (16 bits in the BBC recording you refered) and it should work although first bytes will be rubish but it is not important.

KejPi, 2024-06-15:
Development version of the application supports opening WAV files as raw files. There is not specification for WAV file header for SDR purpose (auxi chunk), closing the issue.

closed 2024-06-15 by KejPi