KejPi/AbracaDABra #142: Wrong service shown than selected shown when scrolling too fast

andimik, 2024-06-08:
This issue is similar to one we had in the very early days. It seems to be a timing problem.

When scrolling the list through very fast (here: with arrow down), it may happen, that the selected service is not that one which is shown.

[image]

Here you see a mixture of 3 different Radio stations from 2 muxes. 

You can see that in the video

[Bildschirmaufzeichnung vom 08.06.2024, 15:27:39.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/5d53472d-a43d-4c2d-92a1-6546ae5817c6)

KejPi, 2024-06-16:
Please check if this issue can be still reproduced with last build.

andimik, 2024-06-16:
No, not fixed.

[image]

KejPi, 2024-06-16:
I do not know how to reproduce it. I have fixed what I could reproduce a since then it is no longer happening on my machine 🙁

andimik, 2024-06-16:
Is there a caching problem with SPI or RadioDNS?

Because the service name now is the right one.

And it seems that there is a check in between as there is a short break and only then the display refreshes.

KejPi, 2024-06-16:
The problem happens when there is a tune to new ensemble, I was not able to reproduce any problem while scrolling in the services from one ensemble.
So in the screenshot Fantasi is selected in the tree, it is indicated by service name, it is playing but the logos belong to VAL 202, correct?
Are the logos displayed as expected when you select the service by "normal" clicking? Actually, are there any logos available for Fantasi service? If not, maybe it could be the reason to be checked.

andimik, 2024-06-17:
[Bildschirmaufzeichnung vom 17.06.2024, 07:19:56.webm](https://github.com/KejPi/AbracaDABra/assets/24510556/f0b92b40-8d39-46f7-858b-18942f1dcb33)

Yes, this solved it.

commit 9b530f06 (2024-06-16): fix: Another attempt to fix #142
closed 2024-06-17 by andimik