KejPi/AbracaDABra #143: stack smashing detected: Segfault when station is currently not transmitting

andimik, 2024-06-28:
[2024-06-28_205022_6A.zip](https://github.com/user-attachments/files/16035647/2024-06-28_205022_6A.zip)

Today I got one of the rare moments when a service (in this case `#Radio ONE` and even `ERF Süd`) is currently not working on an ensemble at all. So I wanted to record its audio (to analyze), but then there is a seg fault. The attached raw file is short (3 seconds), but long enough to see the crash.

```
20:51:12.841 [I] RawFileInput: RAW-FILE: End of file
20:51:15.073 [I] AudioRecorder: Recording file: "/home/andreas/Musik/2024-06-28-205115_E0AD2A_#Radio ONE.aac"
*** stack smashing detected ***: terminated
Abgebrochen (Speicherabzug geschrieben)
```

```
20:57:09.526 [I] AudioRecorder: Recording file: "/home/andreas/Musik/2024-06-28-205709_E0AD24_ERF Süd.aac"
*** stack smashing detected ***: terminated
Abgebrochen (Speicherabzug geschrieben)
```

[image]

KejPi, 2024-06-28:
Thanks for reporting but I cannot reproduce the problem. I can confirm that two services do not play because but how to reproduce the crash using the file you have shared?

KejPi, 2024-06-28:
EDIT: Thanks for reporting, I can reproduce the issue.

KejPi, 2024-06-28:
I have found the root cause, library needs to be updated. 
Note: recording would not help to analyse the problem in this case since the stream is all zero and thus it cannot be decoded.

KejPi, 2024-09-12:
Fixed in 2.5.90

closed 2024-09-12 by KejPi