KejPi/AbracaDABra #144: Transparent PNG background color

andimik, 2024-07-02:
Do you have an idea how to prevent this behavior with transparent PNG files in **SPI**?

[image]
[image]
[image]
[image]
[image]

When **SLS** has been received, then the file is a JPG.
[image]

[image]

KejPi, 2024-07-03:
Well, it is black because all slides have black background that is not visible for non-transparent images but it appears when image is smaller or if it has transparency. I was looking for some guidelines in documentation from WorldDAB and I have not found any recommendation what to do with transparent logos. When I was implementing it, my assumption was that typical car radio has a black screen and dark UI. So I would assume broadcasters should provide their media assets in the format that looks good in car because it is the majority of radio listening. The other assumption that shall be considered is that in dark mode the image background should be dark too. 
To be honest, I do not have any reference how the manufacturers are implementing station logos visualization for DAB, if you have any, please share.

andimik, 2024-07-03:
But the smaller file (top left) is visible, although it is transparent (see files below), too!

So you should use the same background color in the SLS window.

KejPi, 2024-07-03:
Yes, it is visible because I do not set any background there, but try how visible it is in dark mode.

KejPi, 2024-07-04:
This is probably the best compromise. By the way - you can see how stupid they are :-D This is clearly problem on their side,  if they want to have white background they should set it in the logo or make the logo in the way that looks good on any background color in gray scale.

[image]

andimik, 2024-07-07:
BTW: have a look at the letter `R` ...

KejPi, 2024-07-07:
Yes, this is what I meant when calling them stupid. It is also in A letter.

andimik, 2024-09-03:
I just saw that the transparent logos look good in the EPG window, as your background color is the same grey.

[image]

[image]
[image]

KejPi, 2024-09-03:
Try to change to dark mode ;-)

KejPi, 2024-10-19:
@andimik What about this? 
I could implement a possibility to set slideshow background from settings dialog. Would it resolve this issue?

KejPi, 2024-10-24:
User can select any SLS background in settings, closing issue.

closed 2024-10-24 by KejPi