KejPi/AbracaDABra #146: Added EPG button

KejPi, 2024-07-13:
User input: I daily read the EPG (programm guide) und would like a new button for quicker access.

Comment: Options I can see feasible due to space restrictions are:

1) Replace CatSLS button by EPG
2) Add keyboard shortcut for opening EPG
3) Implement a logic to hide buttons under elide (three dots) when they cannot fit
4) Use second line for buttons

closed 2024-08-03 by KejPi

andimik, 2025-03-11:
Just for your information. AbracaDABra also works great with the very detailed Norvegian EPG:

[image]

If you wanna play with it: See https://github.com/user-attachments/files/19195238/EPG_NRK1_abracadabra.zip and https://github.com/user-attachments/files/19194889/raw-raw--Tue-Mar-11-18-14-44-2025.zip

KejPi, 2025-03-11:
Thanks, I like that :-)
FYI: new version is going to show SPI loading progress as you suggested :-)