KejPi/AbracaDABra #147: High memory usage on macOS with version 2.5.1

LuckVintage, 2024-08-10:
Hello there,

I'm using AbracaDABra version 2.5.1 (Apple Silicon) on macOS 14.6.1. I've not had any problems with the app, but recently, I've noticed that it uses a lot of memory, in this case, 2.59 GB (please see screenshot).
[image]

When I open the app and tune in to a station, the memory usage starts small, usually just over 100 MB. After 1 minute, it jumped to just over 1 GB; a minute later, it is over 2.5 GB. It then seems to stabilise somewhere in the 2.5-2.6 GB range. The app continues to work as expected.

A few weeks ago, while I was listening to a station, AbracaDABra consumed as much memory as was available and maxed out my Mac's 8 GB. This hasn't happened again, but I thought I would mention it in case it was related.

I decided to downgrade to version 2.5.0, and this issue went away, with the app not using more than 130 mb, even when using the app for a long period (6 hours).

Many Thanks.

KejPi, 2024-08-11:
Thank you for reporting. It is actually very interesting. There is only minor change in 2.5.1 that only affects RTL-TCP and new RTL-SDR driver. Are you using RTL-TCP? 
I will check if I can reproduce the behaviour on mu Mac but currently I cannot see that the rootcause could be. 
BTW: if you write me privately on my email, I can share development version, if you want to try.

LuckVintage, 2024-08-11:
No worries. I am indeed using RTL-TCP.

I'm more than happy to give the development version a try - where can I find your email address?

KejPi, 2024-08-12:
I am running version 2.5.1 on macOS 14.5 (Apple Silcon) since the morning, monitoring the memory usage and it is stable around 175MB all the time. I am connected to RTL-TCP server running on other machine. I have also checked the changes I did and I do not see anything that might explain behavior you can see :-(

My email address in the "About" dialog.

LuckVintage, 2024-08-12:
Hi again,

Thank you for the development version. Sadly I'm afraid to say the issue does persist, using 2.58 GB memory within 2 minutes as described before.

[image]

Downgraded to release [2.5.0](https://github.com/KejPi/AbracaDABra/releases/tag/v2.5.0), and we're back to using just over 100 MB.

KejPi, 2024-08-12:
I need some test case to reproduce the behavior to fix it. It is working with no issue on two laptops I can use for testing, this would indicate problem caused by reception signal. I can try to update macOS but I do not believe it is the root cause. Is it reproducible using recorded file? Please describe in detail what your use case is. Do you have SPI application enabled? Do you have RadioDNS feature enabled? If this is the case, does it happen when it is disabled? Does it happen with RTL-SDR connected to the computer directly?

LuckVintage, 2024-08-12:
I recorded a 70-second sample with release 2.5.1 and played it with the same version. The memory still increases while using the recorded file, but it doesn't go as high. While recording, the memory went to 2.5 GB like before, but when playing the recorded file, it only went up to 372 MB.

I also played the same recorded file using release 2.5.0, and the memory was fine, not exceeding 123 MB.

I have both the SPI Application and RadioDNS settings enabled. When I disabled RadioDNS, the memory didn't go above 109 MB! When I enabled RadioDNS, the memory started to increase again.

I plugged an RTL-SDR stick directly into my Mac, and the memory also increased like it did with RTL-TCP.

It seems like the memory increase has something to do with having RadioDNS enabled on version 2.5.1.

KejPi, 2024-08-13:
This is very interesting observation and very appreciated, thank you! It could be due to different version of Qt I used to build the packages. AbracaDABra 2.5.0 was build with 6.6.2 while I have built all later builds with 6.5.3 LTS. It is strange that I cannot reproduce it, I will update one of my laptops to latest macOS to see if it makes any difference. By the way - could you please share the recording (by email) so that I can try with the same signal?

LuckVintage, 2024-08-13:
You're welcome, happy to help!

I have just emailed you the same 70-second sample from my tests last night, I hope this is helpful.

KejPi, 2024-08-14:
The rootcause seems to be Qt 6.5.3. When the same source code is built with 6.7.2 the memory consumption is as expected.

closed 2024-08-14 by KejPi