KejPi/AbracaDABra #148: appImage does not detect rtl-sdr stick

andimik, 2024-08-11:
# Problem 

A fresh Debian based installation (like Linux Mint) will not recognize the RTL-Stick with the **appImage** v2.5.1

[image]

```
12:10:23.704 [I] RtlSdrInput: Found 1 devices

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
12:10:23.732 [C] RtlSdrInput: Opening rtl-sdr failed
```

```
lsusb
[...]
Bus 007 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
[...]
```

```
LANG=C && sudo apt-get install rtl-sdr 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rtl-sdr is already the newest version (2.0.1-2build1).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
```

```
LANG=C && sudo apt-get install librtlsdr-dev 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librtlsdr-dev is already the newest version (2.0.1-2build1).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
```

```
LANG=C && sudo apt-get install librtlsdr2 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librtlsdr2 is already the newest version (2.0.1-2build1).
librtlsdr2 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
```

```
sudo dmesg | grep rtl
[ 3431.018043] rtl2832 16-0010: Realtek RTL2832 successfully attached
[ 3431.031017] rtl2832_sdr rtl2832_sdr.1.auto: Registered as swradio0
[ 3431.031034] rtl2832_sdr rtl2832_sdr.1.auto: Realtek RTL2832 SDR attached
[ 3431.031040] rtl2832_sdr rtl2832_sdr.1.auto: SDR API is still slightly experimental and functionality changes may follow
[ 3431.041464] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
```

# Solution

You need to compile it (branch `main`), then it works

```
12:44:31.817 [I] RtlSdrInput: Found 1 devices
Detached kernel driver
Found Rafael Micro R820T tuner
12:44:32.364 [I] RtlSdrInput: Opening rtl-sdr device #0: Generic RTL2832U OEM
[R82XX] PLL not locked!
12:44:32.560 [I] RtlSdrInput: Supported gain values 29
12:44:32.725 [I] RtlSdrInput: Setting bandwidth 1530 kHz
```

[image]

KejPi, 2024-08-11:
I have updated RTL-SDR driver to latest version (I am using version by [old-dab](https://github.com/old-dab/rtlsdr) for all binary releases because it has better performance). There were other users reporting issues with that version, the problem you can see seems to be due to using of the device by DVB-T drivers that need to be blacklisted (Kernel driver is active... message).

KejPi, 2024-08-12:
What about AppImage version 2.5.0? Does it have the same problem?

andimik, 2024-08-12:
Will try it in the next days

andimik, 2024-08-13:
I just tried 2.5.0 AppImage and it works as expected.

```
09:20:12.052 [I] RtlSdrInput: Found 1 devices
Detached kernel driver
FIR Filter 2400 kHz
Found Rafael Micro R820T tuner
09:20:12.553 [I] RtlSdrInput: Opening rtl-sdr device #0: Generic RTL2832U OEM
09:20:12.651 [I] RtlSdrInput: Supported gain values 16
```

KejPi, 2024-08-13:
Thank you for the test. It seesm that latest version of the driver has some issues. I would suppose that blacklisting on unloading DVB-T kernel modules would solve the problem for 2.5.1. I had to copy this file https://github.com/old-dab/rtlsdr/blob/master/rtlsdr-blacklist.conf to `/etc/modprobe.d/` on one of the Ubuntu installations I used for testing it. 
Next time I will try to buid the driver with `DETACH_KERNEL_DRIVER=ON` configuration parameter to see if it helps. But I was never setting it before.

old-dab, 2024-08-23:
You set it before. Andimik's log shows it: "Detached kernel driver".

I startet AppImage 2.5.1 in the terminal. It shows a bit more text.

```
09:51:11.514 [I] RtlSdrInput: Found 1 devices

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
09:51:11.517 [C] RtlSdrInput: Opening rtl-sdr failed

```

KejPi, 2024-08-23:
Thanks for the analysis. I do not remember I have built the driver with any explicit parameter but maybe I am wrong. In any case next time I will use the cmake parametr and I hope it will work as expected. If I ever release new version 🙁

KejPi, 2024-09-12:
[AppImage 2.5.90](https://github.com/KejPi/AbracaDABra/releases/download/v2.5.90/AbracaDABra-v2.5.90-x86_64.AppImage) shoud have RTLSDR driver compiled with detaching. Could you please check it?

andimik, 2024-10-04:
Tested with AppImage 2.5.92 and it works now :)

[image]

closed 2024-10-04 by andimik