KejPi/AbracaDABra #151: Ensemble information/Application log windows  don't behave as SNR/TII windows

Tomfr06, 2024-09-12:
It's minor, but I noticed that both the TII and the SNR windows remember their sizes & positions after restarting the application... 
...but the Ensemble Information and the Application log windows don't.
It would be nice to be have them remember too.
(v2.5.90 on macOS 14.6.1.)

KejPi, 2024-09-12:
You are right, thanks to fo the hint. 
BTW: Do you have Intel or Apple Silicon Mac? I am just curious if anybody uses Intel version of the app.

KejPi, 2024-09-12:
If you use the app on that device than please report any issue you have. I do not want to ask you to be a tester, testing by using it is fine.

KejPi, 2024-09-13:
I have decided to leave behaviour of Settings and Audio recording schedule dialogs as it was so far. Other dialogs should remember their positions and sizes.

closed 2024-09-13 by KejPi