KejPi/AbracaDABra #153: Source code fails to compile on Arch Linux x86-64 due to TII Dialog function error

tghe-retford, 2024-09-16:
The same failure to compile happens with both the beta release in the AUR and the latest main branch code from GitHub. The relevant console output is below:

```
[ 98%] Linking CXX executable AbracaDABra
/usr/bin/ld: CMakeFiles/AbracaDABra.dir/tii/tiidialog.cpp.o: in function `TIIDialog::reset()':
tiidialog.cpp:(.text+0x5d7c): undefined reference to `QCPGraph::setData(QList<double> const&, QList<double> const&, bool)'
/usr/bin/ld: tiidialog.cpp:(.text+0x5e3b): undefined reference to `QCPGraph::setData(QList<double> const&, QList<double> const&, bool)'
collect2: error: ld returned 1 exit status
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:1626: gui/AbracaDABra] Error 1
make[1]: *** [CMakeFiles/Makefile2:305: gui/CMakeFiles/AbracaDABra.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

KejPi, 2024-09-16:
I assume it is because you have installed qcustomplot but not qcustomplot-qt6 from AUR. Either both must be installed or none, I do not know how to deal with this limitation :-(

tghe-retford, 2024-09-16:
Installing qcustomplot-qt6 did allow it to compile. It seems an odd limitation but I guess moving it from optdepends to depends in the PKGBUILD will ensure that qcustomplot-qt6 is installed so no-one else has the same issue. Thank you for your prompt help.

KejPi, 2024-09-16:
Yes, I will probably do it. Thanks for your feedback, I will close this issue.

closed 2024-09-16 by KejPi