KejPi/AbracaDABra #157: AbracaDABra 2.5.92 (RC)  - RTL SDR stick

WIHDIA, 2024-09-22:
Hi Petr,
I've been using AbracaDABra for a long time and today I updated to 2.5.92 because of the TIIs on the map.
Two issues.
1. When turning my stacked antennas on 12B to the nearby strong transmitter, no TIIs are generated on the map unless I toggle a bit
with Gain control (software, device, manual). Just switching a bit and I have TIIs, but they disappear again after a few seconds.
No problem on 12A when the antennas are turned in another direction.
2. Of course I've entered my coordinates and got the FMLIST update. However once having updated I cannot update again.
Perhaps I'm missing something? As Belgium editor of the fmlist I know how often TII changes are implemented and software should be updated. 

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)

KejPi, 2024-09-22:
Hello Herman,
First of all thank you for your blogpost about AbracaDABra, very appreciated!

Now to problems you are facing.  Let's start with 2) that is easy - application does not allow to update TX database more often than once a day. I have implemented this limitation to void server overloading. Do you think it is a serious limitation for overall usability?

Problem number 1) however is more complex. To be honest I have implemented TII detection using my local muxes and few recordings kindly provided by users who participated testing. The only theory I have at the moment is that due to tuner overload there are some artefacts that influence TII detection negatively. It would help if you could share IQ recording of non-working case so that I can do some analysis - I think that about 30 sec would be enough, although longer is better. Please share it privately on my mail that you can find in About dialog of the app. Thank you!

WIHDIA, 2024-09-22:
Hi Petr,

> Hello Herman, First of all thank you for your blogpost about AbracaDABra, very appreciated!

**You're welcome.**
> 
> Now to problems you are facing. Let's start with 2) that is easy - application does not allow to update TX database more often than once a day. I have implemented this limitation to void server overloading. Do you think it is a serious limitation for overall usability?

**That's what I thought what was going on. It's very considerate of you to do so. On the other hand QIRX and Qt-DAB don't do this.  Would Günter object to an unlimited number of daily updates via AbracaDABra?**
> Problem number 1) however is more complex. To be honest I have implemented TII detection using my local muxes and few recordings kindly provided by users who participated testing. The only theory I [...]

**It gets weirder now. When I first noticed this problem it occurred again and again even after restarting AbracaDABra many times.
Then I had to leave home and shut down my pc. Upon returning home and booting my pc I couldn't reproduce the 'error'.
Now I'm stunned :-). I've just made a 12A recording and I'll mail you. Mind you there's no error in this recording.
Perhaps it was just a weird glitch.**

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)

KejPi, 2024-09-22:
For FMLIST update I was proactive implementing this limitation, I did not get specific instructions from Günter. I made a proposal that was accepted. Honestly I do not think this is a major limitation and you can overcome it easily if you really want ;-) 

It seems like signal conditions changed while you were out. Did you try with other app during the "glitch"? I am pretty sure my TII algo is not perfect and you will probably find other glitches. In such case I would appreciate any recording you might want share. 

By the way - my understanding was that 12B was wrong while 12A was OK.

WIHDIA, 2024-09-22:
Hi,

I was in error in the first post. It was indeed 12A.

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)

KejPi, 2024-09-22:
It really looks like some kind of overload.

closed 2024-09-25 by KejPi