KejPi/AbracaDABra #158: SDR play support soon ?

m1ctk, 2024-09-22:
All working nicely but wonder when sdr play devices will be added ?

KejPi, 2024-09-22:
Well, I do not have any SDR Play device. I would be difficult to add support them "blindly" without having the HW  :-(

m1ctk, 2024-09-22:
Yes sure but there are some second hand ones on ebay I see or maybe you can borrow one etc. There are very very popular now amoung sdr users

KejPi, 2024-09-23:
If someone offers to lend me his device for the time necessary to implement support for that in the application I will do it.

m1ctk, 2024-10-08:
> If someone offers to lend me his device for the time necessary to implement support for that in the application I will do it.

But can you not afford to buy one at least second hand  ?!  Well worth it. I doub anyone would lend you one imo

KejPi, 2024-10-09:
Is it not about whather I can afford it or not. I do not want to invest mora than 100 EUR to purchase a HW just to be able to add support for it in application that I give for free. And on top of that I doubt it is worth the money for static DAB reception, there is only marginal benefit compared to cheap RTL-SDR USB stick.

m1ctk, 2024-12-22:
> Is it not about whather I can afford it or not. I do not want to invest mora than 100 EUR to purchase a HW just to be able to add support for it in application that I give for free. And on top of [...]

Well that's shame as the rsp1a and the rsp dx are far more sensitive and selective than a RTL SDR in my experience when used with   Qirx. Hoping someone can lend you one for you to add support

KejPi, 2024-12-22:
Well, it should be significantly better considering the price.
Have you tried with SoapySDR? AbracaDABra supports SoapySDR and SDR Play is supported https://github.com/pothosware/SoapySDRPlay3/wiki

m1ctk, 2024-12-22:
> Well, it should be significantly better considering the price. Have you tried with SoapySDR? AbracaDABra supports SoapySDR and SDR Play is supported https://github.com/pothosware/SoapySDRPlay3/wiki

All to complex to set up. I hope you can simply add sdr play support eventually

closed 2025-03-11 by KejPi