KejPi/AbracaDABra #160: Scanning tool

WIHDIA, 2024-09-25:
Hi,

With the adding of maps and TII decoding AbracaDABra in expert mode has become a bit of DAB DX tool like QIRX or Qt-DAB.
I'm missing though, unless I overlooked it, a real scanning tool with logging. It should work in Expert mode only and of course not
delete the favourites which were already set and should still be there when not in Expert mode. 

73

Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)

KejPi, 2024-09-25:
Hi Herman, 
primary goal is to have an application that used for radio listening, with an easy to use interface and low CPU usage. So I am adding features keeping this target in mind. 

I assume that you only want to create a kind of TXT log from scan without creating service list out of it, correct? 

Just FYI:
Currently there is a possibility to configure the app to not delete service list on band scan. If is is set, it will only update existing items in the list (new ensemble on the same channel) and add new. In ti not possible to have more ensembles in the same channel. Favorites (orange star) are not deleted.  

There is also a possibility to run the application with different settings file. This is typically used by users with variable antenna orientation, e.g. one settings file for orientation west and the other for orientation east.

Please be more specific about the "Scanning tool" features and expected behavior so that I can think about possible implementation of such tool.

WIHDIA, 2024-09-25:
**Hi Petr,**

> Hi Herman, primary goal is to have an application that used for radio listening, with an easy to use interface and low CPU usage. So I am adding features keeping this target in mind. 

**I think that is an excellent target to set.**

> I assume that you only want to create a kind of TXT log from scan without creating service list out of it, correct?

 **Correct.**
 
> Just FYI: Currently there is a possibility to configure the app to not delete service list on band scan. If is is set, it will only update existing items in the list (new ensemble on the same [...]

**I must be missing something as I cannot find that option.**

> There is also a possibility to run the application with different settings file. This is typically used by users with variable antenna orientation, e.g. one settings file for orientation west and the other for orientation east.

**OK. Again how to do this?**
 
> Please be more specific about the "Scanning tool" features and expected behavior so that I can think about possible implementation of such tool.

**This tool should run in Expert Mode only (so non expert users won't see it). Scanning should be done independently from the existing services or favourites list. What should be logged? Timestamp, EID, Ensemble name, TII and if possible connect the TII to location, distance and bearing. Quite a lot :-)**

**73**

**Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)**

KejPi, 2024-09-25:
**I must be missing something as I cannot find that option**
https://github.com/KejPi/AbracaDABra?tab=readme-ov-file#expert-settings

WIHDIA, 2024-09-25:
**Hi Petr,**

> **I must be missing something as I cannot find that option** https://github.com/KejPi/AbracaDABra?tab=readme-ov-file#expert-settings

**Thanks. I should have studied your Github a bit better :-)**
> 
> EDIT: Yes, it is quite a lot but feasible in theory in some of coming releases. Do you have any example of log file? How is the multiple TII logged?

**Do you have QIRX or Qt-DAB? These programs produces logfiles you might want to have a look at.**

**73**

**Herman Wijnants
[My logblog](https://wijnants.info/fmdx/)
Belgium editor of [fmlist.org](https://www.fmlist.org/)**

closed 2024-12-20 by KejPi