KejPi/AbracaDABra #162: show all station logos

sronline, 2024-09-27:
its great that station logos are decoded form SPI :-)

It would be great if all station logos could be displayed:
- 32x32
- 112x32
- 128x128
- 320x240
- 600x600 (RadioDNS)

Also it would be fine if there is an indication where the logos come from (SPI or RadioRNS)

andimik, 2024-09-27:
The second idea is good! We could add it to the mouser over text. 

But regarding the pics: We already have them in all sizes, just see the temporary folder on your hard disk.

KejPi, 2024-09-27:
You can store all objects from SPI application to directory you can select. Timestamp is not supported as a tag, but you can see creation time a file size in filesystem. More info is here https://github.com/KejPi/AbracaDABra?tab=readme-ov-file#user-application-data-storage

Main goal of the app is still radio listening, not data collection and logging. I will follow this target also for future.

KejPi, 2026-10-03:
Timestamp has been added to data storage tokens in 4.3.0