KejPi/AbracaDABra #163: show log with anouncements

sronline, 2024-09-27:
great would be a log of the announcements with date, time, type, length, ....

Perhaps adding a new form which can be opened by clicking the station logo.

KejPi, 2024-09-27:
What would it be good for?

andimik, 2024-10-02:
@sronline 

There is already a log with time and type on the console, so I don't understand your request.

[image]

````
09:02:00.561 [I] RadioControl: "Traffic News" announcement in subchannel 2 cluster ID 1
09:02:12.963 [I] SlideShowApp: Storing slide: "/home/andreas/Downloads/AbracaDABra/SLS/e0ad63/3832sls_3832.jpg"
09:02:31.168 [I] SPIApp: 15: MOT directory NOT complete (decoded 36 / 61)
09:02:32.973 [I] SPIApp: 15: MOT directory NOT complete (decoded 37 / 61)
09:02:35.965 [I] SPIApp: 15: MOT directory NOT complete (decoded 38 / 61)
09:02:40.165 [I] SPIApp: 15: MOT directory NOT complete (decoded 39 / 61)
09:02:46.759 [I] SPIApp: 15: MOT directory NOT complete (decoded 40 / 61)
09:02:48.967 [I] SPIApp: 15: MOT directory NOT complete (decoded 41 / 61)
09:02:51.365 [I] SPIApp: 15: MOT directory NOT complete (decoded 42 / 61)
09:02:55.962 [I] RadioControl: End of announcement
```

KejPi, 2026-10-03:
Closing due to inactivity

closed 2026-10-03 by KejPi