KejPi/AbracaDABra #164: option to delete cache at startup

nona00, 2024-09-28:
for analyzing EPG it would be great to have an option to delete the cache at startup of the application

andimik, 2024-09-28:
Why don't you use a script to delete it for you?

KejPi, 2024-10-06:
Good question!
@nona00 any comment?

nona00, 2024-10-06:
yes this would be possible, but settings in the app to define the cache age for slideshow and spi data would be fine

KejPi, 2024-10-06:
Slideshow has no cache, only SPI data is cached. This includes service logos and EPG. I agree with @andimik that you should delete cache by script if you want, I do not think that this is a feature that others want. I might consider implementing this feature as command line switch - would it fit your use case?

nona00, 2024-10-06:
yes, a command line switch would be great

andimik, 2024-10-07:
I don't think this feature will have more than one user.

andimik, 2024-10-07:
But if you really want that feature for analysis of the SPI, I recommend you to store all SPI files to the drive, see settings menu. Each time, the content changes, you will see a new folder, and you still can browse the old folder and its content.

KejPi, 2024-10-07:
That is very good point!
@nona00 what exactly do you want to do?

nona00, 2024-10-07:
We had some issues with ODR Bridge EPG creator. For different iq recordings I want to check if all logos are received correctly and not that cached data is used. I already enabled storing of SPI files, but then I have to clear cache and download folder.

If nobody else needs this feature I can also create a batch for me.

KejPi, 2024-10-07:
Using SPI data storage is anyway better approach because this way all SPI objects are stored (all logos, binary encoded xml files and decoded version of it and anything else that might be broadcasted in MOT directory) while application stores in its cache only logos 32x32 and 320x240.

andimik, 2024-10-07:
Yes, @KejPi is right. I recommend to save SPI.

closed 2024-10-24 by KejPi