KejPi/AbracaDABra #166: RTLSDR detection fails at cold start

andimik, 2024-10-17:
I don't know the reason, but for me, RTLSDR detection fails after a cold start. I'm using the built version.

[image]

But `lsusb` shows it:

```
lsusb | grep RTL
Bus 007 Device 002: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
```

```
sudo dmesg | grep RTL
[    2.104068] usb 7-1: Product: RTL2838UHIDIR
```

It only works when I unplug it and re-plug in 

[image]

```
sudo dmesg | grep RTL
[    2.104068] usb 7-1: Product: RTL2838UHIDIR
[ 1923.731946] usb 5-2: Product: RTL2838UHIDIR
```

Will further investigate, but wanted to share this issue with you.

KejPi, 2024-10-17:
So it is not detecting device that was connected to the PC before you start the app, correct?
Does any rtl-sdr tool (rtl-tcp for example) see it?

andimik, 2024-10-18:
I'll close, seems to be a problem on my machine only. Right after start the temperature of the RTL stick rises very rapidly.

You were right, even rtl-sdr-tools fail, although I see it in `lsusb` and `dmesg`

```
rtl_test 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_claim_interface error -6
Failed to open rtlsdr device #0.
```

Will also test other USB ports at cold start.