KejPi/AbracaDABra #167: Timing problem with Ensemble label

andimik, 2024-10-17:
There is a tiny timing problem, just cosmetic. Tested with 2.6.0 (built from sources).

When you switch to a DAB signal (here 10D) with the drop down and chance the channel to an empty one (without any signal, here 10C), it may happen that the Ensemble label (here: "SLO DAB+ R1") remains in the header instead of showing nothing:

[image]

Ensemble information is empty of course:

[image]

KejPi, 2024-10-17:
I could reproduce the issue but not anymore after [840316a](https://github.com/KejPi/AbracaDABra/commit/840316a8e1a5672812aa4358e62406f7d3beda72) 
Could you please check?

andimik, 2024-10-18:
Seems this has solved it, thanks!

closed 2024-10-18 by andimik