KejPi/AbracaDABra #173: Program can't scan ensembles if using the second RTL-SDR USB device

beermad, 2024-11-04:
A bit of a strange one here. I have two USB dongles, as I use welle-cli to record programmes and often want them from multiple multiplexes.

If I start AbracaDABra when there isn't a welle-cli process running, it claims the first dongle and can successfully scan for multiplexes. But if welle-cli is running on the first device, AbracaDABra claims the second dongle but fails to scan successfully for multiplexes. If a scan has already been done with the first dongle, then I restart it with the second one, the result is variable. Sometimes It just fails to connect and tells me "no service", on other occasions, the multiplex list loses some or all multiplexes.

I've tried swapping the dongles round in case there's something different about them, but this makes no difference. I've attached logs from AbracaDABra from scanning with the first (dongle1.log) and second (dongle2.log) devices. There seem to be clear differences which I hope are helpful.
[dongle1.log](https://github.com/user-attachments/files/17618771/dongle1.log)
[dongle2.log](https://github.com/user-attachments/files/17618775/dongle2.log)

Other than this, I have to say I'm very happy with this program, which does its job very nicely. Thanks.

KejPi, 2024-11-04:
According to your logs it seems you are running the app under Linux. I have tried to reproduce the issue but I do not see any problem with AbracaDABra running as second. Also the messages in your log look like driver problem. Could you please share more details, specifically:

1. What Linux are you running?
2. Do you use AbracaDABra AppImage or have you built from source?
3. Is it happening with welle.io specifically or with any application using RTL-SDR (like `rtl_test` for example) before starting AbracaDABra?

KejPi, 2025-01-13:
Closing due to inactivity.

closed 2025-01-13 by KejPi