KejPi/AbracaDABra #179: macOS version requirements

krzysztofjeziorny, 2024-11-19:
What specific macOS version is required? The description says

>require at least MacOS BigSur (11)

However, I'm on Sonoma 14.7.1 (Intel), and I'm unable to launch the app (crossed circle icon). Is v14 too new, or too old for the newest build?

Thanks!

krzysztofjeziorny, 2024-11-19:
The arm version works on Sequoia 15.1/M3 Pro, so I suppose it's just an Intel build issue.

krzysztofjeziorny, 2024-11-20:
Version 2.6.0 starts and works well, didn't try 2.5.1 but its icon also looks okay.

KejPi, 2024-11-20:
Thnaks for your test. This is actually very interesting. I am not aware about any chnage I did in build process between 2.6.0 and 2.7.0, but i think I have updated my Mac to macOS 15. I will review the changes in the evening.

svenfl, 2024-11-21:
I cannot start the newest dmg 2.7.1  the former 2.7.1 worked well
I still use qt6.7.3 from homebrew
Here is my log:
Last login: Thu Nov 21 04:20:58 on ttys000
/Applications/AbracaDABra.app/Contents/MacOS/AbracaDABra ; exit;                
sven@Mac ~ % /Applications/AbracaDABra.app/Contents/MacOS/AbracaDABra ; exit;
dyld[13054]: Symbol not found: _qt_version_tag_6_8
  Referenced from: <C25D6D8F-861A-3570-8EC3-B7FB4C6E2228> /Applications/AbracaDABra.app/Contents/Frameworks/libfmlistInterface.1.dylib
  Expected in:     <5E316915-192D-32AE-B085-7C0986279324> /Applications/AbracaDABra.app/Contents/Frameworks/QtCore.framework/Versions/A/QtCore
zsh: abort      /Applications/AbracaDABra.app/Contents/MacOS/AbracaDABra

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Prozess beendet]

KejPi, 2024-11-21:
Thanks, reverted, next trial in the evening ;-)

KejPi, 2024-11-21:
Binaries were updated again, I hope it will work this time.

svenfl, 2024-11-22:
I can confirm works fine again

KejPi, 2024-11-22:
@krzysztofjeziorny Could you please provide your feedback on the last 2.7.1 binaries updated yesterday? Is the issue resolved or not?

krzysztofjeziorny, 2024-11-23:
@KejPi I can let you know in a week's time first, as I'm travelling without the access to my Intel mac.

krzysztofjeziorny, 2024-12-02:
Your newly complied version 2.7.1 works like a charm on Intel macOS Sonoma (14.7.1). Thank you!

KejPi, 2024-12-02:
Thanks for confirmation!

closed 2024-12-02 by KejPi