KejPi/AbracaDABra #187: glibc 2.38 not found for AppImage for Debian Bookworm

andimik, 2024-12-30:
On a fresh installation of Raspberry Pi OS 64bit (Debian Bookworm) I cannot run the AppImage at all, as there are lots of warnings regarding `glibc` not found.

I've `libc6-dev` version 2.36-9-rpt2-deb12u9 on my system, which is the latest stable version, see https://tracker.debian.org/pkg/glibc .

See probono's comment at https://discourse.appimage.org/t/specify-required-glibc-version-in-appimage/2413/4
 ,  you shall take care about minimum `glibc` version.

KejPi, 2025-01-04:
I am sorry, I forgot to answer this issue. 
Glibc version is AppImage limitation, in general AppImage should be compatible with glibc version equal or higher than the version on build machine. I am building the AppImage for AARCH64 under ArchLinux running in Qemu on my Mac and this has currently glibc 2.39, so 2.39 is the minimum version required to run the AppImage.

andimik, 2025-01-04:
This is bad news for me. I reinstalled my Raspberry 4B with 64bit and hoped to get AbracaDABra running.

KejPi, 2025-01-04:
You can use some more up-to-date linux for RPi, I think I was playing with Manjaro in past and it was working well. I have not found better option than Arch that runs in QEMU on macOS at the time I created the environment. I will try to install Debian and let you know.

KejPi, 2025-01-05:
I have tried to build something that might work, please check PM on Rundfunk Forum. I have to apply several dirty tricks so I ma not sure if it works in all use-cases. Please let me know.

andimik, 2025-01-07:
Wow, I'm impressed. It really works on my RPI4B with Debian Bookworm on kernel 6.6.2 

Even TII Decoder and map are shown.

KejPi, 2025-01-07:
Great! Thanks for your feedback. ARM64 AppImage for next application version will be ready for RP4 :-)

KejPi, 2025-01-13:
AppImage v2.9.1  compatible with RPi4 has been released. Closing issue.