KejPi/AbracaDABra #191: Wrong audio output path with Linux Mint 22.1 Cinnamon "Xia"

M4rc-Andre, 2025-01-21:
Hello!

I upgraded my Linux Mint to 22.1 Xia on Sunday. After that, the sound was only output via my LCD monitor through HDMI.
Today I simply set the entry audioFramework=1 in ./.config/AbracaDABra/AbracaDABra.ini. Bevor this was 0.
Now I can select the audio output in the menu. This audioFramework option should be switchable in the menus. Or have I just not found it?

KejPi, 2025-01-21:
You should be able to select audio output in both cases.
Framework switch is not implemented as settings option yet. Current implementation would require application restart that is annoying. 
Do I understand correctly that you did not have any options to select for audio framework = 0 (PortAudio)?

M4rc-Andre, 2025-01-22:
Hello!

I use AppImage. See Screenshots.

Bevor:

audioDevice=
audioFramework=0

After i set audioFramework=1 and close AbracaDABra:

audioDevice=@ByteArray(alsa_output.usb-Burr-Brown_from_TI_USB_Audio_CODEC-00.analog-stereo-output)
audioFramework=1

[image]

[image]

KejPi, 2025-01-22:
Yes, it is expected behavior. I have check the source code and audio device selection with PortAudio was never supported under Linux. As I wrote before, in that case application should use default audio output configured in the system. 
Lets keep this issue open as feature request for runtime switching of audio output framework to replace manual editing of the config file.