KejPi/AbracaDABra #193: Software gain works worse than manual gain

andimik, 2025-03-05:
I don't know what is the culprit, but when I build the software from sources under Linux, the software gain is worse than the manual gain. This is on all active channels I can receive here.

For example, this video, I'm just switching between sw and manual gain on the same antenna (signal is stable).

https://github.com/user-attachments/assets/c1a16812-f808-4e48-9506-7807c0c415ef

So I made a comparison:

channel | software gain (dB) | manual gain (dB)
---| --- |--
5A | 4* | 14
5B | 4* | 15
6A | 0* | 11
6B | 10 | 21
6D |0* | 16 
10D | 6 | 12
12B | 4* | 5*
12C | 4* | 10

 \* means no audio

This is Linux 

```
System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble

CPU:
  Info: quad core model: AMD A8-4500M APU with Radeon HD Graphics bits: 64 type: MT MCP
    smt: enabled arch: Piledriver rev: 1 cache: L1: 192 KiB L2: 4 MiB
  Speed (MHz): avg: 1389 high: 1397 min/max: 1400/1900 boost: enabled cores: 1: 1391 2: 1395
    3: 1397 4: 1375 bogomips: 15173
  Flags: avx ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm

```

KejPi, 2025-03-05:
Does it happen with AppImage too?

KejPi, 2025-03-05:
Is there anything interesting in terminal output?

andimik, 2025-03-05:
No, nothing which might help. It first locks, plays an audio for around 1 second, but then immediately loses the audio:

`AudioDecoder: Muting audio (decoding errors)`

When I change to manual gain, the console output is of course

`AudioDecoder: Umuting audio`

KejPi, 2025-03-05:
I do not know what could be wrong, it works as expected on my side. The only difference I can identify is that I am using RTL-SDR driver by old-dab and you are probably using the one from your linux installation. But it is probably not the rootcause because manual AGC settings work.

andimik, 2025-03-05:
Initially I was testing a02a1328a0a13f614dbd7cec1f6a9cd30cd54fc1 which is v2.9.0

Now I'm on branch v2.9.2 with `git checkout v2.9.2` But with the default RTL driver it's the same issue.

Then I had the idea to use `rtl_tcp` instead, this works now much better as the manual gain now is nearly identical, except for two channels.

channel | sw | man
-- | -- | --
5A | 14 | 15
5B | 14 | 15 
6A | 20 | 20
6B | 12 (!) | 22
6D | 25 |  25 
10D | 10 | 12
12B | 5* | 5*
12C  | 5* | 11

KejPi, 2025-03-06:
Please share IQ recording from RTL-SDR with following contents: 10 seconds of signal with good manual AGC settings, then switch to SW AGC a continue the recording for next 15 seconds. 

And what is the SW AGC doing with the gain? Is it oscillating? You can see current gain value in Ensemble dialog.

andimik, 2025-03-06:
Well, all SW gain experiments have one interesting thing:

The SNR starts good (green) but drops down within 1 or 2 seconds.

Ok, will record them and share the spectrum. Good idea.

KejPi, 2025-03-06:
One additional idea - probably unrelated, but worth to try - could you please try to extend USB buffer:

`sudo bash -c 'echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb'`

Then unplug and plug USB device.

andimik, 2025-03-11:
Due to a hardware failure I had to remove the amplifier. The funny thing is, that it works without the amplifier.

That's why I don't have news here. Still need to test.

But the buffer commands were not helping me.

KejPi, 2025-03-11:
Thanks for reply, amplifier might explain the problems with SW AGC.

andimik, 2025-03-11:
Maybe you are right. Shall I close it?

KejPi, 2025-03-11:
If you think you may provide some more information (recording etc.) in short time, then keep it open, otherwise I would suggest to close it for now, you can reopen it later.

closed 2025-06-03 by KejPi