KejPi/AbracaDABra #195: Idea: Show country flag (ECC)

ryanginn, 2025-03-16:
Hi, this is a great program! Thank you for making it 🙂 

A suggestion I just thought about would be the ability to show the country flag of the station; maybe beside the station name when tuned. All services are able to have unique ECCs transmitted over the air, and the multiplex has it's own as shown when you hover over the ensemble name.

Please let me know if you want an IQ sample of a multiplex which has this!

Rhían Mac G

andimik, 2025-03-16:
Please note, that there are restrictions on various hardware multiplexers. That's why the ECC is not always correct. For example Israel in Italy.

KejPi, 2025-03-16:
I can display a flag somewhere but not in main window. I do not think this information is interesting for average user. I will think about that.

@andimik I am curious, do you have a short recording of the Italian ensemble pretending to be Israel?

andimik, 2025-03-16:
I really apologize, I don't have access to my DAB archive at the moment, but I just found a very old file from 2012, which is also showing Israel.

```
Ensemble:
0x512D RASDAB12D [ RASDAB ] ECC: 0xE0, UTC 120 min, INT: 0, alarm announcements: 0
Services (8):
0x42F3 RAS CH-JAZZ [ CH-JAZZ ] ECC: 0xE0, Country: **Israel**, PTy: Jazz Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS CH-JAZZ' [ 'CH-JAZZ' ], ASCTy: 0x0 (MP2)
      SubChId: 8, Language: Unknown/NA, StartCU: 752, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x43A1 RAS R. RUMANTSCH [ RAS RR ] ECC: 0xE0, Country: **Israel**, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS R. RUMANTSCH' [ 'RAS RR' ], ASCTy: 0x0 (MP2)
      SubChId: 6, Language: Unknown/NA, StartCU: 544, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
0xA201 RAS OE 1 [ RAS OE 1 ] ECC: 0xE0, Country: Austria, PTy: Culture (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS OE 1' [ 'RAS OE 1' ], ASCTy: 0x0 (MP2)
      SubChId: 1, Language: Unknown/NA, StartCU: 0, NumCU: 116, UEP #40, Protection level: 3, Bitrate: 160kbps
0xA203 RAS OE 3 [ RAS OE 3 ] ECC: 0xE0, Country: Austria, PTy: Pop Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS OE 3' [ 'RAS OE 3' ], ASCTy: 0x0 (MP2)
      SubChId: 3, Language: Unknown/NA, StartCU: 232, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xA213 RAS FM 4 [ RAS FM 4 ] ECC: 0xE0, Country: Austria, PTy: Other Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS FM 4' [ 'RAS FM 4' ], ASCTy: 0x0 (MP2)
      SubChId: 4, Language: Unknown/NA, StartCU: 336, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
0xAA02 RAS OE 2 Tirol [ RAS OE 2 ] ECC: 0xE0, Country: Austria, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS OE 2 Tirol' [ 'RAS OE 2' ], ASCTy: 0x0 (MP2)
      SubChId: 2, Language: Unknown/NA, StartCU: 116, NumCU: 116, UEP #40, Protection level: 3, Bitrate: 160kbps
0xD311 RAS BAYERN 1 [ RAS BAY1 ] ECC: 0xE0, Country: Germany, PTy: Varied (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS BAYERN 1' [ 'RAS BAY1' ], ASCTy: 0x0 (MP2)
      SubChId: 5, Language: Unknown/NA, StartCU: 440, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
0xD312 RAS BAYERN 2 [ RAS BAY2 ] ECC: 0xE0, Country: Germany, PTy: Culture (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS BAYERN 2' [ 'RAS BAY2' ], ASCTy: 0x0 (MP2)
      SubChId: 7, Language: Unknown/NA, StartCU: 648, NumCU: 104, UEP #39, Protection level: 4, Bitrate: 160kbps
```

https://we.tl/t-sb4GDCRPf2 (link valid only a few days till 19th March)

As far as I remember, the technician told me that it is **not possible to mix ECC in an ensemble** (not sure if hardware limited or spec violation otherwise), so they needed to stay on 0xE0 for all services. Switzerland is 0xE1 4xxx and Israel is 0xE0 4xxx 

For example http://poupa.cz/rds/countrycodes.htm

**But it's still the case**, see current log on https://www.fmlist.org/urds/urdslister_dabsvc.php?id=19359&block=10B&eid=510b

[image]

ryanginn, 2025-03-16:
> As far as I remember, the technician told me that it is not possible to mix ECC in an ensemble (not sure if hardware limited or spec violation otherwise), so they needed to stay on 0xE0 for all services. Switzerland is 0xE1 4xxx and Israel is 0xE0 4xxx

It is possible to mix ECC in an ensemble. I have such in my ODR multiplex. It is limited however by the SID of course, same with FM ECC.

> I can display a flag somewhere but not in main window. I do not think this information is interesting for average user. I will think about that.

That would be great! Thanks! 🙂

andimik, 2025-03-16:
I'm a bit curious that you have achieved it, because I have **never seen** a mux with mixed ECC 

https://www.fmlist.org/urds/urdslister_dabsvc.php?id=19357&block=12A&eid=5003

[image]

I found quite a lot muxes from Italy and Switzerland where ECC is the same for all services.

https://www.fmlist.org/urds/urdslister_dabsvc.php?id=19463&block=7A&eid=4201

[image]

Therefore, there are also services on air which seem to be from Jordan or Libya.

So, please, don't expect that this function will show the truth.

KejPi, 2025-03-17:
@andimik Thank you for sharing the recording. I hope no DAB transmission like that is still on air. There is a possibility to signalise correct ECC for each service individually - see example below where 3 countries are signalised correctly. 

```
Ensemble:
0x510B RAS1-DAB [ RAS1-DAB ] ECC: 0xE0, UTC 60 min, INT: 1, alarm announcements: 0
Services (11):
0x5201 Rai Radio1 TAA [ Rai1 TAA ] ECC: 0xE0, Country: Italy, PTy: N/A, Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Rai Radio1 TAA' [ 'Rai1 TAA' ], ASCTy: 0x3F (AAC)
      SubChId: 1, Language: Unknown/NA, StartCU: 624, NumCU: 48, EEP 3-A, Coderate: 1/2, Bitrate: 64kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x5202 Rai Radio2 TAA [ Rai2 TAA ] ECC: 0xE0, Country: Italy, PTy: N/A, Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Rai Radio2 TAA' [ 'Rai2 TAA' ], ASCTy: 0x3F (AAC)
      SubChId: 2, Language: Unknown/NA, StartCU: 672, NumCU: 48, EEP 3-A, Coderate: 1/2, Bitrate: 64kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x5203 Rai Radio3 [ Radio3 ] ECC: 0xE0, Country: Italy, PTy: N/A, Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'Rai Radio3' [ 'Radio3' ], ASCTy: 0x3F (AAC)
      SubChId: 3, Language: Unknown/NA, StartCU: 720, NumCU: 72, EEP 3-A, Coderate: 1/2, Bitrate: 96kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x5404 RAI SÜDTIROL [ RAI SÜDT ] ECC: 0xE0, Country: Italy, PTy: N/A, Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAI SÜDTIROL' [ 'RAI SÜDT' ], ASCTy: 0x3F (AAC)
      SubChId: 4, Language: Unknown/NA, StartCU: 0, NumCU: 48, EEP 3-A, Coderate: 1/2, Bitrate: 64kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xD220 RAS Dlf Kultur [ RAS DKul ] ECC: 0xE0, Country: Germany, PTy: Culture (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS Dlf Kultur' [ 'RAS DKul' ], ASCTy: 0x3F (AAC)
      SubChId: 9, Language: Unknown/NA, StartCU: 366, NumCU: 84, EEP 3-A, Coderate: 1/2, Bitrate: 112kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xD313 RAS BAYERN 3 [ RAS BAY3 ] ECC: 0xE0, Country: Germany, PTy: Pop Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS BAYERN 3' [ 'RAS BAY3' ], ASCTy: 0x3F (AAC)
      SubChId: 5, Language: Unknown/NA, StartCU: 48, NumCU: 84, EEP 3-A, Coderate: 1/2, Bitrate: 112kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xD314 RAS BR-KLASSIK [ RAS BR-K ] ECC: 0xE0, Country: Germany, PTy: Serious Classical (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS BR-KLASSIK' [ 'RAS BR-K' ], ASCTy: 0x3F (AAC)
      SubChId: 6, Language: Unknown/NA, StartCU: 132, NumCU: 90, EEP 3-A, Coderate: 1/2, Bitrate: 120kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xD315 RAS BR24 [ RAS BR24 ] ECC: 0xE0, Country: Germany, PTy: News (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS BR24' [ 'RAS BR24' ], ASCTy: 0x3F (AAC)
      SubChId: 7, Language: Unknown/NA, StartCU: 222, NumCU: 72, EEP 3-A, Coderate: 1/2, Bitrate: 96kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0xDF95 RAS die Maus [ RAS Maus ] ECC: 0xE0, Country: Germany, PTy: Children's programmes (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS die Maus' [ 'RAS Maus' ], ASCTy: 0x3F (AAC)
      SubChId: 8, Language: Unknown/NA, StartCU: 294, NumCU: 72, EEP 3-A, Coderate: 1/2, Bitrate: 96kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x42F1 RAS Swiss Pop [ RAS SPop ] ECC: 0xE1, Country: Switzerland, PTy: Pop Music (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS Swiss Pop' [ 'RAS SPop' ], ASCTy: 0x3F (AAC)
      SubChId: 11, Language: Unknown/NA, StartCU: 540, NumCU: 84, EEP 3-A, Coderate: 1/2, Bitrate: 112kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
0x42F2 RAS SwissClassic [ RAS SCla ] ECC: 0xE1, Country: Switzerland, PTy: Light Classical (static), Announcements: No
AudioComponent (primary), SCIdS: 0, Label: 'RAS SwissClassic' [ 'RAS SCla' ], ASCTy: 0x3F (AAC)
      SubChId: 10, Language: Unknown/NA, StartCU: 450, NumCU: 90, EEP 3-A, Coderate: 1/2, Bitrate: 120kbps
            UserApp 1/1: Label: '' [ '' ], UAType: 0x2 (SlideShow), X-PAD AppTy: 12, DSCTy: 0x3C (MOT), DG: 0, Data (2) [0C3C]
```

andimik, 2025-03-17:
No, you are wrong. There are still at least 5 muxes with such a constellation on air. All of them do NOT change the ECC.

ryanginn, 2025-03-17:
How is he wrong? In the excerpt sent by @KejPi it is clearly shown the countries of Switzerland Italy and Germany....

andimik, 2025-03-17:
> I hope no DAB transmission like that is still on air. 

There he is wrong.

KejPi, 2025-03-18:
In any case it is a problem on brooadcasting side, nothing to worry about from application point of view.

ryanginn, 2025-04-13:
> I'm a bit curious that you have achieved it, because I have **never seen** a mux with mixed ECC
>
> https://www.fmlist.org/urds/urdslister_dabsvc.php?id=19357&block=12A&eid=5003
>
> [image]
>
> I found quite [...]

It is indeed possible, please see attatched

[2025-04-14_001523_12C_MPBNL DAB+.csv](https://github.com/user-attachments/files/19727793/2025-04-14_001523_12C_MPBNL.DAB%2B.csv)

closed 2025-07-03 by KejPi