KejPi/AbracaDABra #198: 3.1 won't run with MacOS Sequoia - Can't verify

m1ctk, 2025-03-28:
As title. no go with me, anyone else ?

KejPi, 2025-03-28:
It runs, I have just verified on AppliSilicon MacBook runnin last version of Sequoia. Are you runnning it on Intel?

If it is about the message from attached screenshot, then it is normal behavior of the application that is not signed and notarized. You need to click Done, then go to Settings -> Privacy & Security and enable the application if you trust it. If you don't, then you cannot run the app, sorry.

[image]

m1ctk, 2025-03-28:
Oh yes thanks 'open anyway' worked however it doesn't see my rspdx though SDRconnect does ?

KejPi, 2025-03-28:
Do you have SDRplay API 3.15 installed and daemon running?

m1ctk, 2025-03-28:
> Do you have SDRplay API 3.15 installed and daemon running?

I presume I must have otherwise SDR connect wouldn't see the dx ? I don't know what daemon is ?

KejPi, 2025-03-28:
I would say yes, nevertheless I have just tried and it works. I can connect to RSP1A and it works fine.

[image]

m1ctk, 2025-03-28:
Not with me or the 1a, intel here btw on a 2012 mac mini using OpenCore

KejPi, 2025-03-28:
I cannot test the app on Intel Mac but I can use RSP1A when I run Intel version on Apple Silicon Mac. So it shoud work in general. If you run the application from terminal is there anything intersting in the log that is printer to terminal output?

m1ctk, 2025-03-29:
All good now, just needed the API installed !

closed 2025-03-30 by KejPi