KejPi/AbracaDABra #204: AppImage aarch64 AppRun.wrapped

andimik, 2025-04-02:
I just had several reproducible crashes, when deleting the service list and scanning again. It's the AppImage for aarch64.

> AppRun.wrapped: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
> Aborted

(BTW: mind the wrong character in front of pthread ... :-)

```
20:30:59.267 [I] RadioControl: Tuning 0.000 -> 174.928 MHz ...
AppRun.wrapped: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.

Thread 1 "AppRun.wrapped" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=549620966592, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (threadid=549620966592, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x0000007ff4f30aa4 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000007ff4eea72c in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000007ff4ed747c in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000007ff4ee41a4 in __assert_fail_base (fmt=0x7ff4ffef38 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7ff4e93378 "pthread_mutex_lock(mutex) == 0", 
    file=file@entry=0x7ff4e93358 "../../libusb/os/threads_posix.h", line=line@entry=46, function=function@entry=0x7ff4e97700 "usbi_mutex_lock") at ./assert/assert.c:92
#5  0x0000007ff4ee420c in __GI___assert_fail (assertion=0x7ff4e93378 "pthread_mutex_lock(mutex) == 0", file=0x7ff4e93358 "../../libusb/os/threads_posix.h", line=46, function=0x7ff4e97700 "usbi_mutex_lock") at ./assert/assert.c:101
#6  0x0000007ff4e8f02c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#7  0x0000007ff4e91c9c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#8  0x0000007ff4e9271c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#9  0x0000007ff4e8abc8 in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#10 0x0000007ff4e8c09c in libusb_handle_events_timeout_completed () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#11 0x0000007ff4e8c190 in libusb_handle_events_completed () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#12 0x0000007ff4e8cde8 in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#13 0x0000007ff4e8d0bc in libusb_control_transfer () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#14 0x0000007ff7e37a04 in rtlsdr_reset_buffer () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/librtlsdr.so
#15 0x000000555567b0f0 in RtlSdrInput::tune(unsigned int) ()
#16 0x0000007ff63a7a5c in QObject::event(QEvent*) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#17 0x0000007ff74c9120 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Widgets.so.6
#18 0x0000007ff635a6c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#19 0x0000007ff635dd8c in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#20 0x0000007ff65e0e18 in ?? () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#21 0x0000007ff45e774c in g_main_context_dispatch () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libglib-2.0.so.0
#22 0x0000007ff45e79e0 in ?? () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libglib-2.0.so.0
#23 0x0000007ff45e7a84 in g_main_context_iteration () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libglib-2.0.so.0
#24 0x0000007ff65e06b4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#25 0x0000007ff6365710 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#26 0x0000007ff6362084 in QCoreApplication::exec() () from /tmp/.mount_Abraca8LxUVD/usr/bin/../lib/libQt6Core.so.6
#27 0x000000555559bf34 in main ()
```

```
$ uname -a
Linux rpi3b 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

andimik, 2025-04-02:
```
$ apt list libusb*
Listing... Done
libusb-0.1-4/stable,now 2:0.1.12-32 arm64 [installed,automatic]
libusb-0.1-4/stable 2:0.1.12-32 armhf
libusb-1.0-0-dev/stable,now 2:1.0.26-1 arm64 [installed]
libusb-1.0-0-dev/stable 2:1.0.26-1 armhf
libusb-1.0-0/stable,now 2:1.0.26-1 arm64 [installed]
libusb-1.0-0/stable 2:1.0.26-1 armhf
libusb-1.0-doc/stable,stable,now 2:1.0.26-1 all [installed,automatic]
libusb-dev/stable,now 2:0.1.12-32 arm64 [installed]
libusb-dev/stable 2:0.1.12-32 armhf
libusb-java-doc/stable,stable 0.8+ztex20090101-9 all
libusb-java-lib/stable 0.8+ztex20090101-9 arm64
libusb-java-lib/stable 0.8+ztex20090101-9 armhf
libusb-java/stable,stable 0.8+ztex20090101-9 all
libusb-libusb-perl/stable 0.09-2+b3 arm64
libusb-libusb-perl/stable 0.09-2+b3 armhf
libusb-ocaml-dev/stable 1.3.1-4+b11 arm64
libusb-ocaml-dev/stable 1.3.1-4+b11 armhf
libusb-ocaml/stable 1.3.1-4+b11 arm64
libusb-ocaml/stable 1.3.1-4+b11 armhf
libusb3380-0/stable 0.0.1+git20190125.c83d1e9-2 arm64
libusb3380-0/stable 0.0.1+git20190125.c83d1e9-2 armhf
libusb3380-dev/stable 0.0.1+git20190125.c83d1e9-2 arm64
libusb3380-dev/stable 0.0.1+git20190125.c83d1e9-2 armhf
libusbauth-configparser-dev/stable 1.0.5-2 arm64
libusbauth-configparser-dev/stable 1.0.5-2 armhf
libusbauth-configparser1/stable 1.0.5-2 arm64
libusbauth-configparser1/stable 1.0.5-2 armhf
libusbguard-dev/stable 1.1.2+ds-3+b1 arm64
libusbguard-dev/stable 1.1.2+ds-3+b1 armhf
libusbguard1/stable 1.1.2+ds-3+b1 arm64
libusbguard1/stable 1.1.2+ds-3+b1 armhf
libusbgx-dev/stable 0.2.0-2 arm64
libusbgx-dev/stable 0.2.0-2 armhf
libusbgx-doc/stable,stable 0.2.0-2 all
libusbgx2/stable 0.2.0-2 arm64
libusbgx2/stable 0.2.0-2 armhf
libusbmuxd-dev/stable 2.0.2-3 arm64
libusbmuxd-dev/stable 2.0.2-3 armhf
libusbmuxd-tools/stable 2.0.2-3 arm64
libusbmuxd-tools/stable 2.0.2-3 armhf
libusbmuxd6/stable,now 2.0.2-3 arm64 [installed,automatic]
libusbmuxd6/stable 2.0.2-3 armhf
libusbredirhost-dev/stable 0.13.0-2 arm64
libusbredirhost-dev/stable 0.13.0-2 armhf
libusbredirhost1/stable 0.13.0-2 arm64
libusbredirhost1/stable 0.13.0-2 armhf
libusbredirparser-dev/stable 0.13.0-2 arm64
libusbredirparser-dev/stable 0.13.0-2 armhf
libusbredirparser1/stable 0.13.0-2 arm64
libusbredirparser1/stable 0.13.0-2 armhf
```

1.0.26 should be latest for my system, although 1.0.28 is available on GIthub.

also found this:
https://github.com/libusb/libusb/issues/1605

KejPi, 2025-04-03:
Thanks for reporting. 
I have tried several times today (at least 5x times) to clear service list and run band scan and it did not crash. I used RTL-SDR device for this test. 

When exactly is it crashing? 
Do you use RTL-SDR? 
Do you do anything more than just Menu->Clear service list->OK->Menu->Band scan?

andimik, 2025-04-10:
Yes, RTL SDR.

No, I get the crash when starting a new band scan.

Before, I can listen to all muxes.

andimik, 2025-05-30:
**I tried rtl_tcp, and this works as expected**. Even AbracaDABra is as fast as I know from a PC.

With the internal rtl driver, the app was quite slow.

andimik, 2025-06-01:
Well, now the app is as fast as with an Airspy and does not produce fake reception (tuner works), but there is still the crash when I want to clear the Service List.

When I plug in the Airspy, no crash with "Clear Service List".

KejPi, 2025-06-01:
I have tried to rebuild it with libusb 1.0.28, not sure if it was successful, could you please try: https://www.uschovna.cz/zasilka/SUAAXFBU3RMUTPLV-E7X
By the way - I remember that was similar issue with fake ensembles on Linux in past but I do not remember what was the fix. But I would say the problem was not in the application.

andimik, 2025-06-04:
No changing, still crashing and duplicates. 

[image]

andimik, 2025-06-05:
No, the power supplier is not the culprit.

With version 04 the gdb output is:

```
AppRun.wrapped: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
[Thread 0x7fb584ebc0 (LWP 12887) exited]

Thread 1 "AppRun.wrapped" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=549620967424, signo=signo@entry=6, 
    no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (threadid=549620967424, signo=signo@entry=6, 
    no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x0000007ff4f30ab4 in __pthread_kill_internal (signo=6, 
    threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x0000007ff4eea72c in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
#3  0x0000007ff4ed747c in __GI_abort () at ./stdlib/abort.c:79
#4  0x0000007ff4ee41b8 in __assert_fail_base (
    fmt=0x7ff4ffef38 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7ff4e93378 "pthread_mutex_lock(mutex) == 0", 
    file=file@entry=0x7ff4e93358 "../../libusb/os/threads_posix.h", 
    line=line@entry=46, function=function@entry=0x7ff4e97700 "usbi_mutex_lock")
    at ./assert/assert.c:94
#5  0x0000007ff4ee421c in __GI___assert_fail (
    assertion=0x7ff4e93378 "pthread_mutex_lock(mutex) == 0", 
    file=0x7ff4e93358 "../../libusb/os/threads_posix.h", line=46, 
    function=0x7ff4e97700 "usbi_mutex_lock") at ./assert/assert.c:103
#6  0x0000007ff4e8f02c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#7  0x0000007ff4e91c9c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#8  0x0000007ff4e9271c in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#9  0x0000007ff4e8abc8 in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#10 0x0000007ff4e8c09c in libusb_handle_events_timeout_completed ()
   from /lib/aarch64-linux-gnu/libusb-1.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#11 0x0000007ff4e8c190 in libusb_handle_events_completed () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#12 0x0000007ff4e8cde8 in ?? () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#13 0x0000007ff4e8d0bc in libusb_control_transfer () from /lib/aarch64-linux-gnu/libusb-1.0.so.0
#14 0x0000007ff7e33774 in ?? () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/librtlsdr.so
#15 0x0000007ff7e34a34 in rtlsdr_set_tuner_gain_index () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/librtlsdr.so
#16 0x0000007ff7e34b6c in rtlsdr_set_tuner_gain () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/librtlsdr.so
#17 0x0000005555688784 in RtlSdrInput::setGain(int) ()
#18 0x0000007ff63a7a5c in QObject::event(QEvent*) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#19 0x0000007ff74c9120 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Widgets.so.6
#20 0x0000007ff635a6c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#21 0x0000007ff635dd8c in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#22 0x0000007ff65e0e18 in ?? () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#23 0x0000007ff450766c in g_main_context_dispatch () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libglib-2.0.so.0
#24 0x0000007ff4507900 in ?? () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libglib-2.0.so.0
#25 0x0000007ff45079a4 in g_main_context_iteration () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libglib-2.0.so.0
#26 0x0000007ff65e06b4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#27 0x0000007ff6365710 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#28 0x0000007ff6362084 in QCoreApplication::exec() () from /tmp/.mount_AbracaGgTV1u/usr/bin/../lib/libQt6Core.so.6
#29 0x000000555559d2b4 in main ()
```

The crash is not immediately. Clearing works, but selecting any channel (5A in my case) will for sure crash.

andimik, 2025-06-05:
No, never, as I thought aarch image is still experimental.

On the same Raspi I am regularly scanning with other tools, but none of them produce fake reception with rtl stick.

For example my current (running) log

```
scanEvalDABens.sh          7,"10D",0x9200,"SLO DAB+ R1     "
3,"12B",0x9300,"SLO DAB+ R2 E   "
7,"12C",0x9400,"SLO DAB+ R2 W   "
7,"5A",0x9600,"SLO DAB+ R4     "
7,"6A",0xa101,"DAB+ Austria    "
7,"6B",0xa202,"DAB+ MUX 3-StBK "
7,"6D",0xa304,"DAB+ MUX 2-StBK "
1,"8D",0xa304,"DAB+ MUX 2-StBK "
```

KejPi, 2025-06-05:
I got RPi4 for a test and I do not see any problem. I have prepared SD card with PI OS 64 bit, I updated the packages after the boot, I did not install any additional package. I blacklisted DVB-T, applied udev rules so that I can use the device as regular user (both config files downloaded from https://github.com/old-dab/rtlsdr). Then I rebooted the system, increased USBFS buffer after reboot and started the app. It did band scan, there were no fake ensembles and application works as expected. I do not see any problem in the RPi 4 I have. I used RTL-SDR V3 plugged in blue USB port.

andimik, 2025-06-05:
Oh, I see. As I might be the only user under aarch64, I will use rtl_tcp. 

So let's close this before it gets too time consuming for both of us ...

closed 2025-06-05 by andimik