KejPi/AbracaDABra #206: Soapy SDR plugin path when not used

andimik, 2025-04-18:
https://github.com/KejPi/AbracaDABra/blob/3d7d07bdcbc0cb1d685917193fcf3d0a86d5045c/gui/main.cpp#L55

Although I have not used Soapy in my cmake, I cannot build the app as there is an error in this line.

```
 31%] Building CXX object gui/CMakeFiles/AbracaDABra.dir/main.cpp.o
/home/andreas/apps/abracadabra/gui/main.cpp: In function ‘int main(int, char**)’:
/home/andreas/apps/abracadabra/gui/main.cpp:55:46: error: invalid initialization of reference of type ‘const QByteArray&’ from expression of type ‘QByteArrayView’
   55 |             qputenv("SOAPY_SDR_PLUGIN_PATH", QByteArrayView(QString(QCoreApplication::applicationDirPath() + "/../lib/SoapySDR/").toUtf8()));
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtGui/qtguiglobal.h:7,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/qtwidgetsglobal.h:7,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/qapplication.h:7,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/QApplication:1,
                 from /home/andreas/apps/abracadabra/gui/main.cpp:27:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qglobal.h:1356:67: note: in passing argument 2 of ‘bool qputenv(const char*, const QByteArray&)’
 1356 | Q_CORE_EXPORT bool qputenv(const char *varName, const QByteArray& value);
      |                                                 ~~~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [gui/CMakeFiles/AbracaDABra.dir/build.make:401: gui/CMakeFiles/AbracaDABra.dir/main.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:173: gui/CMakeFiles/AbracaDABra.dir/all] Fehler 2
```

I also tried to use `#ifdef HAVE_SOAPYSDR` but it did not work.

The only workaround is to comment out this line 55, then this file at least is build (but I still cannot build it to the end).

My settings (Ubuntu 24.04 LTS, Linux Mint)

```
 AIRSPY                           ON
 AIRSPY_INCLUDE_DIRS              /usr/include
 AIRSPY_LINK_LIBRARIES            /usr/lib/x86_64-linux-gnu/libairspy.so
 APPLE_APP_BUNDLE                 OFF
 APPLE_BUILD_X86_64               OFF
 CMAKE_BUILD_TYPE
 CMAKE_INSTALL_PREFIX             /usr/local
 GIT_SUBMODULE                    ON
 LIBFAAD2_INCLUDE_DIRS            /usr/include
 LIBFAAD2_LINK_LIBRARIES          /usr/lib/x86_64-linux-gnu/libfaad.so
 LIBMPG123_INCLUDE_DIRS           /usr/include/x86_64-linux-gnu
 LIBMPG123_LINK_LIBRARIES         /usr/lib/x86_64-linux-gnu/libmpg123.so
 LIBUSB_LINK_LIBRARIES            /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 PORTAUDIO_LINK_LIBRARIES         /usr/lib/x86_64-linux-gnu/libportaudio.so
 QCUSTOMPLOT                      OFF
 QCUSTOMPLOT_INCLUDE_DIRS         QCUSTOMPLOT_INCLUDE_DIRS-NOTFOUND
 QCUSTOMPLOT_LINK_LIBRARIES       QCUSTOMPLOT_LINK_LIBRARIES-NOTFOUND
 QT_ADDITIONAL_HOST_PACKAGES_PR   
 QT_ADDITIONAL_PACKAGES_PREFIX_   
 QT_DIR                           /usr/lib/x86_64-linux-gnu/cmake/Qt6
 Qt6CoreTools_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6CoreTools
 Qt6Core_DIR                      /usr/lib/x86_64-linux-gnu/cmake/Qt6Core
 Qt6DBusTools_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6DBusTools
 Qt6DBus_DIR                      /usr/lib/x86_64-linux-gnu/cmake/Qt6DBus
 Qt6GuiTools_DIR                  /usr/lib/x86_64-linux-gnu/cmake/Qt6GuiTools
 Qt6Gui_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Gui
 Qt6LinguistTools_DIR             /usr/lib/x86_64-linux-gnu/cmake/Qt6LinguistTools
 Qt6Multimedia_DIR                /usr/lib/x86_64-linux-gnu/cmake/Qt6Multimedia
 Qt6Network_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt6Network
 Qt6OpenGL_DIR                    /usr/lib/x86_64-linux-gnu/cmake/Qt6OpenGL
 Qt6Positioning_DIR               /usr/lib/x86_64-linux-gnu/cmake/Qt6Positioning
 Qt6QmlCompilerPlusPrivate_DIR    Qt6QmlCompilerPlusPrivate_DIR-NOTFOUND
 Qt6QmlIntegration_DIR            /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlIntegration
 Qt6QmlModels_DIR                 /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlModels
 Qt6QmlTools_DIR                  /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlTools
 Qt6QmlWorkerScript_DIR           /usr/lib/x86_64-linux-gnu/cmake/Qt6QmlWorkerScript
 Qt6Qml_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Qml
 Qt6QuickControls2_DIR            /usr/lib/x86_64-linux-gnu/cmake/Qt6QuickControls2
 Qt6QuickTemplates2_DIR           /usr/lib/x86_64-linux-gnu/cmake/Qt6QuickTemplates2
 Qt6Quick_DIR                     /usr/lib/x86_64-linux-gnu/cmake/Qt6Quick
 Qt6Svg_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Svg
 Qt6WidgetsTools_DIR              /usr/lib/x86_64-linux-gnu/cmake/Qt6WidgetsTools
 Qt6Widgets_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt6Widgets
 Qt6Xml_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt6Xml
 Qt6_DIR                          /usr/lib/x86_64-linux-gnu/cmake/Qt6    
 RARTTCP                          OFF
 RTL_SDR_INCLUDE_DIRS             /usr/local/include
 RTL_SDR_LINK_LIBRARIES           /usr/local/lib/librtlsdr.so      
 SOAPYSDR                         OFF
 TII_SPECTRUM_PLOT                OFF
 USE_FDKAAC                       OFF
 USE_PORTAUDIO                    ON
 USE_SYSTEM_AIRSPY                ON
 USE_SYSTEM_FAAD                  ON
 USE_SYSTEM_FDKAAC                ON
 USE_SYSTEM_LIBUSB                ON
 USE_SYSTEM_MPG123                ON
 USE_SYSTEM_PORTAUDIO             ON
 USE_SYSTEM_QCUSTOMPLOT           ON
 USE_SYSTEM_RTLSDR                ON
 USE_SYSTEM_SOAPYSDR              OFF
 XKB_INCLUDE_DIR                  /usr/include
 XKB_LIBRARY                      /usr/lib/x86_64-linux-gnu/libxkbcommon.so  
```

KejPi, 2025-04-18:
I guess you are building with Qt < 6.5.

andimik, 2025-04-18:
Yes, but I expect that I at least can build it without TII, so I switched it off.

> Note: Currently you can still compile the application with obsolete Qt version delivered with Ubuntu 24.04 but it is an unsupported configuration with limited functionality.

KejPi, 2025-04-18:
Well, you are probably the only one who is trying to build it with obsolete Qt version, Qt 6.5 was released more than 2 years ago. I will fix this for next version.

KejPi, 2025-04-18:
Fixed in devel branch.

closed 2025-04-18 by KejPi