KejPi/AbracaDABra #207: build error: ‘m_signalDialog’ was not declared in this scope

andimik, 2025-04-18:
https://github.com/KejPi/AbracaDABra/blob/3d7d07bdcbc0cb1d685917193fcf3d0a86d5045c/gui/mainwindow.cpp#L4495

This line causes an error:

```
/home/andreas/apps/abracadabra/gui/mainwindow.cpp: In member function ‘virtual void MainWindow::changeEvent(QEvent*)’:
/home/andreas/apps/abracadabra/gui/mainwindow.cpp:4495:17: error: ‘m_signalDialog’ was not declared in this scope; did you mean ‘showSignalDialog’?
 4495 |             if (m_signalDialog && m_signalDialog->isVisible())
      |                 ^~~~~~~~~~~~~~
      |                 showSignalDialog
ninja: build stopped: subcommand failed.
19:56:02: Der Prozess "/usr/bin/cmake" wurde mit dem Rückgabewert 1 beendet.
Fehler beim Erstellen/Deployment des Projekts AbracaDABra (Kit: Desktop)
Bei der Ausführung von Schritt "Erstellen"
```

I tried it with Qt Creator.

I expect that I - at least - can build it with Qt < 6.5 on Ubuntu 24.04.

KejPi, 2025-04-18:
This is already fixed in b0800cfe7abebf79f1edb915b3cf55fe96129017

andimik, 2025-04-18:
I don't think so. I have cloned the recent version, so I did not miss this commit.

```
$ git log | grep b0800cf
commit b0800cfe7abebf79f1edb915b3cf55fe96129017
```

There is something other preventing that.

KejPi, 2025-04-18:
You are right.

KejPi, 2025-04-18:
You cannot compile devel branch because I have not updated DABSDR library yet.

closed 2025-04-18 by KejPi

andimik, 2025-04-18:
Ok, this explains why there are errors in

```
            stats.audioServiceBytes = pData->audioServiceBytes;
            stats.padBytes = pData->padBytes;
```

KejPi, 2025-04-18:
Yes, I should be able to update library next week. I am almost ready for the release.

andimik, 2025-04-18:
Ok, then Happy Easter!

KejPi, 2025-04-18:
Thanks, happy Easter!

andimik, 2025-04-26:
BTW: On a different laptop with Ubuntu 24.10 and Qt 6.6.2 I could build it.

andimik, 2025-04-26:
And after deleting everything in the build-directory I could build it against Qt 6.4.2 (Ubuntu 24.04).

Seems I made a wrong cmake setting.