KejPi/AbracaDABra #211: Antenna switching without disconnecting (for RSPDX)

m1ctk, 2025-04-25:
Please can this be done with a future update as it would really make life easier for those with 2 or more antenna connected !

KejPi, 2025-04-25:
Actually, I tried to implement it for 3.2.0 but the user I asked for testing it concluded it did not work on RSPdx so I removed the implementation for the release. I do not have multi-antenna SDRplay HW, I even had to borrow RSP1A to be able to add support for SDRplay in the application since it is almost impossible to implement such features without having HW for debugging it. If you want to work as tester for this feature I give it another try. Contact me on my email directly if you want to help. My email address is in About dialog.

m1ctk, 2025-04-27:
> Actually, I tried to implement it for 3.2.0 but the user I asked for testing it concluded it did not work on RSPdx so I removed the implementation for the release. I do not have multi-antenna [...]

OK yes understand. I can work as a tester sure if possible. I don't see any about dialog though. My email is amstereofan@gmail.com