KejPi/AbracaDABra #212: Make DLS+ show beside DLS instead of having to scroll

ryanginn, 2025-04-25:
Hi, as of the new version I cannot see the DLS+ and the normal DLS text together, I have to scroll for it. Even increasing the size of the window does not show it, instead it increases the size of the slideshow. Would it be possible to fix this? 😀

See below: I am using Windows 11.

[image]

[image]

ryanginn, 2025-04-26:
> Yes, good idea. Although enlarging the windows works under Linux, then you
> see the DL+.
>
> My proposal is to fix the dimension of the SLS to 320x240 pixels, because
> enlarging the complete main window will add a CinemaScope-like feeling.
>
> grafik.png (view on web)

Yes I agree with you

KejPi, 2025-04-27:
I do not see any problem on Windows 11

[image]

ryanginn, 2025-04-27:
> I do not see any problem on Windows 11
>
> 2025-04-27.20_13_23-Abraca.DAB.Radio.png (view on web)

That is weird… 🤔

KejPi, 2025-04-28:
Did you try just to increase window height?

ryanginn, 2025-04-28:
Yes, I did. That only increases slideshow size, not the DLS+ part

ryanginn, 2025-04-28:
Of course, see below:

[image]

https://youtu.be/QGifoxx7T2U

I can get it all on one page, but it has to be very large

ryanginn, 2025-04-28:
The confusing thing is it wasn't like this before I updated Abracadabra to latest version 🤔 I haven't updated the Windows or anything since upgrading the software...

KejPi, 2025-04-28:
It seems that video was blocked :-( Another try: https://youtube.com/shorts/dcosgxOyNFE?feature=share

KejPi, 2025-04-28:
So if you run 3.1.0 on current system it behaves differently than 3.2.0? The only explanation I have is different W11 version. I will try on another W11 PC tomorrow. 
By the way, just to be sure, are you using binary downloaded from Github not your own build, correct?

KejPi, 2025-04-29:
It works as expected on another W11 installation 

[image]

[image]

ryanginn, 2025-04-29:
Huh weird

ryanginn, 2025-04-29:
> Just brainstorming because I have no idea what is wrong :-(
>
>    - Did you unpack the AbracaDABra 3.2.0 zip file as new directory (no
>    overwriting of old application directory)?
>    - Do you have Qt development environment installed?
>    - When you go to Menu -> About, do you see Qt version 6.7.3 there?
>    - Could you please try to remove (rename) settings INI file and start
>    with empty settings, does it happen too?

Will look when I am home , thanks !

andimik, 2025-04-29:
What is the screen resolution? Does it also happen at 100% zoom?

ryanginn, 2025-04-29:
1920x1080, 100% zoom

KejPi, 2025-04-29:
I have reverted the only change I did with main windows for 3.2.0, could you please try this test build [www.uschovna.cz/zasilka/SI5CAJ5VDS52VP87-X8H](https://www.uschovna.cz/zasilka/SI5CAJ5VDS52VP87-X8H)

ryanginn, 2025-04-29:
Did you unpack the AbracaDABra 3.2.0 zip file as new directory (no overwriting of old application directory)? -> Yes
Do you have Qt development environment installed? - No
When you go to Menu -> About, do you see Qt version 6.7.3 there? - Yes 
Could you please try to remove (rename) settings INI file and start with empty settings, does it happen too? -> there is not one in the directory. I can't find it.

I tried this version you send, and still same thing! 

Logs:

20:04:37.025 [I] Application: Version: v3.2.0-3-g8c88952
20:04:37.110 [W] default: No QtMultimedia backends found. Only QMediaDevices, QAudioDevice, QSoundEffect, QAudioSink, and QAudioSource are available.
20:04:37.159 [I] Application: Using PortAudio output
20:04:37.195 [I] RtlSdrInput: Found 1 devices.
20:04:37.204 [I] RtlSdrInput: #0: Generic RTL2832U OEM | Realtek | RTL2838UHIDIR | 00000001
20:04:37.278 [I] Application: Proxy config: System
20:04:37.293 [I] RtlSdrInput: Found 1 devices
20:04:37.715 [I] RtlSdrInput: Opening rtl-sdr device: Realtek | RTL2838UHIDIR | SN: 00000001
20:04:37.803 [I] RtlSdrInput: Supported gain values 16
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Note FM            E180AA : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Radio Nostalgia    E180AB : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Pacific FM         E1F8D9 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Pacific FM Relax   E1F8E7 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Radio MIX          E23010 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Planeta FM         E23169 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 2 Easy   E32490 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 1        E32491 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 2        E32492 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] riseFM             E32493 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Raidió Rí-Rá       E32494 : 0
20:04:37.836 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] S21 Radio          E32495 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] S21 Euro           E32496 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 1 MixM   E32497 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 2 90s    E32498 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Disco Groove       E32499 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Classic Hits 80s   E32869 : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Queer Hits         E38AAA : 0
20:04:37.837 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Val 202            E49202 : 0
20:04:37.837 [I] ServiceList: Service list loaded: number of services in all ensembles 19, unique services 19
20:04:37.906 [I] RtlSdrInput: Setting bandwidth 1530 kHz resulted to 1550 kHz
20:04:37.914 [I] RtlSdrInput: Found 1 devices.
20:04:37.921 [I] RadioControl: Tuning 0.000 -> 227.360 MHz ...
20:04:38.345 [I] RadioControl: Current frequency 227.360 MHz
20:04:39.190 [W] SPIApp: DNS lookup failed: "0.3010.2e01.3e2.dab.radiodns.org"
20:04:39.199 [W] SPIApp: DNS lookup failed: "0.2869.2e01.2e3.dab.radiodns.org"
20:04:39.204 [W] SPIApp: DNS lookup failed: "0.2492.2e01.2e3.dab.radiodns.org"
20:04:39.245 [W] SPIApp: DNS lookup failed: "0.2496.2e01.2e3.dab.radiodns.org"
20:04:39.249 [W] SPIApp: DNS lookup failed: "0.8aaa.2e01.8e3.dab.radiodns.org"
20:04:39.253 [W] SPIApp: DNS lookup failed: "0.80ab.2e01.8e1.dab.radiodns.org"
20:04:39.258 [W] SPIApp: DNS lookup failed: "0.2499.2e01.2e3.dab.radiodns.org"
20:04:39.261 [W] SPIApp: DNS lookup failed: "0.f8d9.2e01.fe1.dab.radiodns.org"
20:04:39.265 [W] SPIApp: DNS lookup failed: "0.2498.2e01.2e3.dab.radiodns.org"
20:04:39.268 [W] SPIApp: DNS lookup failed: "0.2495.2e01.2e3.dab.radiodns.org"
20:04:39.272 [W] SPIApp: DNS lookup failed: "0.2497.2e01.2e3.dab.radiodns.org"
20:04:39.275 [W] SPIApp: DNS lookup failed: "0.3169.2e01.3e2.dab.radiodns.org"
20:04:39.279 [W] SPIApp: DNS lookup failed: "0.2490.2e01.2e3.dab.radiodns.org"
20:04:39.283 [W] SPIApp: DNS lookup failed: "0.2494.2e01.2e3.dab.radiodns.org"
20:04:39.286 [W] SPIApp: DNS lookup failed: "0.f8e7.2e01.fe1.dab.radiodns.org"
20:04:39.289 [W] SPIApp: DNS lookup failed: "0.80aa.2e01.8e1.dab.radiodns.org"
20:04:39.293 [W] SPIApp: DNS lookup failed: "0.2493.2e01.2e3.dab.radiodns.org"
20:04:39.297 [W] SPIApp: DNS lookup failed: "0.9202.2e01.9e4.dab.radiodns.org"
20:04:39.570 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 1        E32491 : 0
20:04:39.570 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 2        E32492 : 0
20:04:39.570 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] riseFM             E32493 : 0
20:04:39.571 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] Raidió Rí-Rá       E32494 : 0
20:04:39.572 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] S21 Radio          E32495 : 0
20:04:39.572 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] S21 Euro           E32496 : 0
20:04:39.572 [I] ServiceList:           [E32E01 @ 227360 kHz | 12C] MPB Radio 1 MixM   E32497 : 0
[... cut ...]
20:04:40.150 [I] AudioDecoder: HE-AAC 48 kHz stereo
20:04:40.150 [I] AudioDecoder: Output sample rate 48000 Hz, channels: 2
20:04:40.167 [I] AudioOutput: Audio device "Speakers / Headphones (Realtek High Definition Audio)" [default]
20:04:40.751 [I] RadioControl: Starting user application 'SlideShow' from XPAD.
20:04:40.752 [I] Application: Uploading ensemble information to FMLIST (3232 bytes)
20:04:40.790 [I] AudioOutput: Unmuting audio
20:04:41.110 [I] Application: Ensemble information uploaded, thank you!

KejPi, 2025-04-29:
Settings is in `%USERPROFILE%\AppData\Roaming\AbracaDABra\AbracaDABra.ini`
What about this build: https://www.uschovna.cz/zasilka/SI82589YWZCTSKKN-593

ryanginn, 2025-04-29:
Wiping the INI file did nothing, still have to scroll. Will check your new build now.

ryanginn, 2025-04-29:
Nope, new version did not fix; see here: https://www.youtube.com/watch?v=ntdBj21JIwo

KejPi, 2025-04-29:
Could you please share IQ recording of the ensemble? It is very strange, I am running out of ideas but it seems that you are no the only one facing this issue :-( It seems to be dependent on the selected service. Meanwhile I am preparing another test build.

ryanginn, 2025-04-29:
Yes, of course. I will make an IQ recording this moment!

KejPi, 2025-04-29:
Thanks, new test build is here: https://www.uschovna.cz/zasilka/SI92PB68MWX8KIJF-ZMC
I am sorry it is try-and-error approach using you as tester but I really do not know where the problem could be :-(

ryanginn, 2025-04-29:
How can I record IQ in AbracaDABra? I only see "Raw" which doesn't output .raw ?

KejPi, 2025-04-29:
Menu -> Ensemble information -> Record raw data 

https://github.com/KejPi/AbracaDABra?tab=readme-ov-file#ensemble-information

ryanginn, 2025-04-29:
Gotcha, I will record a few minutes and send onto you

ryanginn, 2025-04-29:
https://nc.mpbnl.nl/index.php/s/EgmKYNeAJdXkJWE

Try this, hope it works

KejPi, 2025-04-29:
I cannot reproduce any issue :-( https://youtube.com/shorts/WEM5aOFEQC4

ryanginn, 2025-04-29:
That is so weird!

ryanginn, 2025-04-29:
> I cannot reproduce any issue :-( https://youtube.com/shorts/WEM5aOFEQC4

That is so weird!

KejPi, 2025-04-29:
What about the last test build I have shared? https://www.uschovna.cz/zasilka/SI92PB68MWX8KIJF-ZMC

ryanginn, 2025-04-29:
Still not!

KejPi, 2025-04-29:
OK, I will try to share another test build tomorrow. It is more and more weird 😕

KejPi, 2025-04-30:
One more idea, could you please try this? https://www.uschovna.cz/zasilka/SJM33TX6D7XWWU3Y-ZA4

KejPi, 2025-04-30:
One user on [Runfunk forum](https://www.rundfunkforum.de/viewtopic.php?p=1805135#p1805135) writes this: 

> I restarted the program and switched the Abraca DAB Radio window to full screen and back again, and I can't find any errors now. I can't say what's changed.

I know it sounds weird, but could you try please?

andimik, 2025-04-30:
@ryanginn 

Dear Ryan, thanks for the file. Even when it's lab generated, it confirms that AbracaDABra can handle clickable URLs in Slideshow!

And there are some special characters which might be useful to test with different hard- and software.

ryanginn, 2025-04-30:
> One user on [Runfunk forum](https://www.rundfunkforum.de/viewtopic.php?p=1805135#p1805135) writes this:
>
>
> I know it sounds weird, but could you try please?

Hi, Sorry for the delay. Sadly this did not work for me!

ryanginn, 2025-04-30:
[image]

Nope

KejPi, 2025-04-30:
And this https://www.uschovna.cz/zasilka/SJMIT7JJLXTJ6DNX-2JL

ryanginn, 2025-04-30:
[image]

edd-cgfan, 2025-04-30:
> And this https://www.uschovna.cz/zasilka/SJMIT7JJLXTJ6DNX-2JL

Hi KejPi,

This test version works on Windows 10 like version 3.1.0 did without scrolling, however, one had to enlarge the height of the main window a little bit:

[image]

With a test version earlier I had to increase the main window's height much more:

[image]

I had a look at your mainwindow.ui file:
slsView_Service has a sizePolicy "Expanding" for both Height and Width and no value for maximum or minimum, but dynamicLabel_Announcement has only "Preferred" for both and additionally a maximumSize value set to 100 for the height.

I would change this to "Expanding" for dynamicLabel_Announcement. For the slides it could be "Preferred" or "Fixed" because the slides usually have one size.

Maybe this helps.

Regards,
cgfan

andimik, 2025-04-30:
But the last screenshots look normal as expected.

You have not enlarged the window.

KejPi, 2025-05-02:
The point is that when you increase height of the window, first DL(+) area shall expand and then SLS should start to expand. This is what needs to be tested.

ryanginn, 2025-05-02:
> Ryan, But the last screenshots look normal as expected.
> 
> You have not enlarged the window.

Before, I did not need to enlarge it to see DLS+. That is the problem 🙂

ryanginn, 2025-05-02:
Hmm interesting. I didn't have to before. This must be something to do on my end then 🤔

KejPi, 2025-05-11:
So what is the status? Is any of the builds, I have shared, working as expected? I would like to resolve this issue finally.

edd-cgfan, 2025-05-11:
Hi again KejPi,

Yes, I can confirm that this version works as expected.
Just checked it again and here you can see that enlarged to this height it shows the text without scrolling.

[image]

Thanks for your efforts.

Regards,
cgfan

edd-cgfan, 2025-05-12:
This version does work here on Win10:

[image]

The original version 3.1.0 published earlier does work too here.

KejPi, 2025-05-12:
Thanks for your reply. Stupid question, just to be sure, version 3.2.0 does not work on your Windows, correct?

edd-cgfan, 2025-05-12:
Yes, correct.

andimik, 2025-05-12:
That's strange, I don't see a difference in 3.1 and 3.2

#3.1

[image]

# 3.2
[image]

andimik, 2025-05-12:
and this is the version mentioned above under Windows 10.

# AbracaDABra-v3.1.0-5-g02b4d47.zip

[image]

KejPi, 2025-05-12:
Could you please update your status for remaining test builds? 

1. https://www.uschovna.cz/zasilka/SI5CAJ5VDS52VP87-X8H ... @ryanginn NO | @edd-cgfan **???**
2. https://www.uschovna.cz/zasilka/SI82589YWZCTSKKN-593 ... @ryanginn  NO | @edd-cgfan  **???**
3. https://www.uschovna.cz/zasilka/SI92PB68MWX8KIJF-ZMC ... @ryanginn  NO | @edd-cgfan  **???**
4. https://www.uschovna.cz/zasilka/SJM33TX6D7XWWU3Y-ZA4 ... @ryanginn  NO | @edd-cgfan  **???**
5. https://www.uschovna.cz/zasilka/SJ2PWIYS2XKI8LC8-WCB ... @ryanginn  NO | @edd-cgfan YES
6. https://www.uschovna.cz/zasilka/SJMIT7JJLXTJ6DNX-2JL ... @ryanginn  ??? | @edd-cgfan  YES

edd-cgfan, 2025-05-12:
Please see the difference:
**3.1.0:** The slide does not enlarge with black stripes and text is fully visible:

[image]

**3.2.0** While enlarging the window the slide gets the black areas but text is not fully visible:

[image]

KejPi, 2025-05-12:
Yes, I understand what is wrong with the application behaviour on your Windows. The problem is that it seems to happen on some windows installations only and I cannot reproduce it on 2 Windows installation I can use for checking, neither @andimik  obviously can. And I cannot find in the code any change that would explain different behaviour. That is why I have generated these test builds to isolate what commit caused the issue. I rely on you to tell me, what works and what does not work.

edd-cgfan, 2025-05-12:
Here are my latest test results:

> Could you please update your status for remaining test builds?
> 
> 1. https://www.uschovna.cz/zasilka/SI5CAJ5VDS52VP87-X8H ... [@ryanginn](https://github.com/ryanginn) NO | [@edd-cgfan](https://github.com/edd-cgfan) NOT OK
> 2. https://www.uschovna.cz/zasilka/SI82589YWZCTSKKN-593 ... [@ryanginn](https://github.com/ryanginn)  NO | [@edd-cgfan](https://github.com/edd-cgfan)  NOT OK
> 3. https://www.uschovna.cz/zasilka/SI92PB68MWX8KIJF-ZMC ... [@ryanginn](https://github.com/ryanginn)  NO | [@edd-cgfan](https://github.com/edd-cgfan)  OK
> 4. https://www.uschovna.cz/zasilka/SJM33TX6D7XWWU3Y-ZA4 ... [@ryanginn](https://github.com/ryanginn)  NO | [@edd-cgfan](https://github.com/edd-cgfan)  NOT OK
> 5. https://www.uschovna.cz/zasilka/SJ2PWIYS2XKI8LC8-WCB ... [@ryanginn](https://github.com/ryanginn)  NO | [@edd-cgfan](https://github.com/edd-cgfan) YES
> 6. https://www.uschovna.cz/zasilka/SJMIT7JJLXTJ6DNX-2JL ... [@ryanginn](https://github.com/ryanginn)  ??? | [@edd-cgfan](https://github.com/edd-cgfan)  YES

KejPi, 2025-05-12:
It seems I can reproduce the issue now. 
Please try to set Windows theme to light and start the application or change the theme to light and back to dark while application is running. 
In my case when I change theme to dark before I start the app, it behaves wrong, but changing it to light while application is running fixes the behavior.

edd-cgfan, 2025-05-12:
You made me smile: This procedure really works over here too!
Nice finding.
However, after closing Abra and new start one has to do this again in order to get the wanted behavior.

KejPi, 2025-05-12:
Yes, the same on my side, when app starts while Win is in dark mode, it does not work. When it starts while Windows is in light mode, it works as expected. By the way, I can reproduce wrong behavior with 3.1.0 when started in Windows dark mode too.

edd-cgfan, 2025-05-12:
Yes, the same with 3.1.0 here.

KejPi, 2025-05-12:
Could you pleas etry this build, it works on my W11: [www.uschovna.cz/zasilka/SQXT2NF9EM4HJXKA-2NY](https://www.uschovna.cz/zasilka/SQXT2NF9EM4HJXKA-2NY)

@ryanginn Could you please try it too and report the result?

[5 more replies]