KejPi/AbracaDABra #213: 3.2.0: Ensemble Information Window is cropped

andimik, 2025-04-26:
https://github.com/KejPi/AbracaDABra/blob/3b1be3f3ea960262b7784e1933a05c26761e84dc/gui/ensembleinfodialog.ui#L10

I cannot see the complete Ensemble Information window with 3.2.0, because there is a task bar at the bottom (or top) in some OS.

Please reduce to 672 like before, it will be enlarged anyway to the maximum.

# 3.2.0

[image]

# 3.1.0

[image]

(Code was https://github.com/KejPi/AbracaDABra/blame/727ee0a65faf87a6d34f21cda386307b84486e57/gui/ensembleinfodialog.ui )

KejPi, 2025-04-26:
Just for my curiosity, what is your screen resolution?

andimik, 2025-04-26:
1366x768, which is quite common.

andimik, 2025-04-26:
This is the fix:

https://github.com/KejPi/AbracaDABra/blob/3b1be3f3ea960262b7784e1933a05c26761e84dc/gui/ensembleinfodialog.ui#L1035

should be for example

```
       <height>200</height>
```

[image]