KejPi/AbracaDABra #218: Feature request: Option to manually clear scan results instead of auto-clearing on restart

HansVanEijsden, 2025-05-10:
During a band scan, I sometimes want to stop the scan to listen to a station I'm currently receiving - especially during tropo conditions.

However, when I press Stop and then Start again, the scan results are automatically cleared and the scan starts from scratch. This causes me to lose all previously found entries (see screenshot) unless I remember to export the list as CSV first. Merging multiple CSVs afterward is a hassle, especially during active DX conditions.

Suggestion:
Add a button (e.g. Clear List) to manually clear the results, and let the list persist when stopping and restarting the scan - unless manually cleared by the user.

This would make it much easier to pause for listening without losing valuable data.
Thanks for considering it!

[image]

andimik, 2025-05-10:
Why don't you use the drop down menu and change the channel manually? This will not clear the list.

KejPi, 2025-05-10:
It makes sense, I will implement it, thanks for suggestion.

KejPi, 2025-05-16:
I am actually facing in issue how to integrate the feature in UI (as usual :-( ) because I do not have enough space for new button so I would like to understand the use-case. Is the use case that sometimes you want to clear the results and in some cases not? Or would it make sense to implement it in the way that (if the option is set), the scanner will never delete previous results and it will always add it until you keep the dialog window opened.

HansVanEijsden, 2025-05-17:
Thank you for asking.. and for your patience! I was on the road the past few days.

Your second proposed solution sounds perfect to me: if an option is enabled, the scanner should never clear the previous results and simply keep adding to the list, as long as the Scanning Tool dialog window remains open. That would work very well for my use case, since it allows me to pause, listen to a station in between, and then continue scanning without losing anything.

KejPi, 2025-05-19:
After considering it again I thing that this feature would make most sense for infinite scan. I will probably add it for this case only.

closed 2025-07-03 by KejPi