KejPi/AbracaDABra #220: Feaure Request: TPEG Traffic Information display / dump

nica-f, 2025-05-19:
First of all - Your AbracaDABra is amazing! One of the best DAB+/SDR apps I have seen so far, works like a charm and offers pretty much everything one could dram of, except...

In old FM radio days there were cheap and simple FM radio tuner chips that were able to output the RDS TMC data stream, using location tables one could decode traffic information. With DAB this is now replaced with TPEG.

It would be really really awesome if AbracaDABra could also display or at least dump the TPEG data stream too for further processing - like displaying in some mapping application.

Thank you!

KejPi, 2025-05-20:
TPEG is currently out of scope, I do not know about any available decoder that I could easily integrate in the application and I do not have bandwidth to develop my own. Furthermore I do not have any signal that I can use for testing. I can imagine to implement some stream output containign TPEG data as they are so that it can be fed to some decoder but I need specification for that.

nica-f, 2025-05-20:
Thank you so much for the fast response!
And yes, I am also not aware of any TPEG stream decoders - yet. And I think one of the reasons is that almost nothing exposes the raw data stream so that someone could develop such a decoder.
So as a first step it would be awesome if the raw TPEG data stream could just be dumped to a file. In Germany a couple of muxes carry a TPEG stream (I can copy details for you from your excellent DAB mux info page in the app) and these TPEG streams are also usually not encrypted.
Once I have dumped such a stream I an also provide a copy for hacking of course :-)

edd-cgfan, 2026-10-30:
Hi @KejPi,

Have you noticed what @andimik found and posted on [rundfunkforum.de.](https://www.rundfunkforum.de/viewtopic.php?p=1873001#p1873001):
[dab-radio by rtoenniges](https://github.com/rtoenniges/dab-radio)
This is the first time I see the decoding of DAB-TPEG information.
In order to get an imagination, here is a screenshot:
[image]

andimik, 2026-10-30:
yes, I found it by chance.

Sadly I cannot test it here due to lack of TPEG, and (more sadly) this tool does not accept rawfiles as input and does NOT accept rtl_tcp with rawfiles (I know the trick, but it does not work with this tool).