KejPi/AbracaDABra #222: Suggestion: autolog

WIHDIA, 2025-06-06:
Hi Petr,

A suggestion from a Dutch friend and AbracaDABra user. Wouldn't it be useful if a AbracaDABra user had a choice to set the TII logger and the Scanning tool to a autologger mode so everything popping up on these widgets would be automatically (or manually, as is the case now) logged?

Thanks for your great DAB software!

73
Kind regards

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
Belgium editor [fmlist.org](https://www.fmlist.org/)

WIHDIA, 2025-06-07:
Hi Petr,

No idea, of course, how it should be working. The suggestion was inspired by the scanning tool in QIRX which generates rather small
(up to 4 MB) after a night of continuous scanning. 

73
Kind regards

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
Belgium editor [fmlist.org](https://www.fmlist.org/)

KejPi, 2025-06-07:
I am sorry, I still do not understand the use case.

So you start Scanning tool with endless number of cycles in the evening and in the morning you stop it. And the suggestion is to store the log automatically when you stop it instead of doing it manually?

KejPi, 2025-06-07:
Well, anything can happen.
OK, it is somehow clear for Scanning tool. But you mentioned also TII in the first post. How is that supposed to work? Currently you can open TII dialog and press recording button to let application write the log.

WIHDIA, 2025-06-07:
Hi Petr,

The most interesting part is of course the scanning tool. Perhaps the TII could ask the user to save the codes found after closing the widget? Then again I personally don't often record and save codes from the TII widget as I prefer using the scanning tool.

73
Kind regards

Herman Wijnants
[My logblog](https://www.wijnants.info/fmdx/)
Belgium editor [fmlist.org](https://www.fmlist.org/)