KejPi/AbracaDABra #230: Ussie with "dab-tx-list.csv"

gvanem, 2025-07-28:
I'm building Abracadabra on Win-10, but have problems with retrieving and using this `dab-tx-list.csv` file.
So I get no distances to my local transmitters. How is this supposed to be downloaded?

But I made this GNU-make snippet to download it:
```
TII_CACHE = $(realpath $(LOCALAPPDATA))/AbracaDABra/cache/TII

get_dab_list:
	mkdir --parents $(TII_CACHE)
	curl -sko $(TII_CACHE)/dab-tx-list.csv https://www.dablist.org/qtdab/qtdab_dabtx_data.php
```

Works fine now (my home is at the blue dot):

[image]

But I do not understand what this `Main 23, Sub 5` with no `Distance` and `Azimuth` is for. 

*BTW*. It is very confusing with all the different directories in Abracadabra.
&nbsp; &nbsp;  &nbsp;   &nbsp; IMHO it should be enough with **one** directory-branch under `$(APPDATA)` or `$(HOME)`. Why so many?

gvanem, 2025-07-28:
> TII database is updated from Settings -> TII -> Transmitter database -> Update

That is greyed out when `dab-list.csv` does not exist:

[image]

Hence my doubt about the CMake settings or some other missing requirement.

KejPi, 2025-07-28:
It looks like this when database is not available. 
[image]

It is disabled when FMLIST interface library is not available.

gvanem, 2025-07-28:
> It is disabled when FMLIST interface library is not available.

Well, since FMLIST interface library is made with MinGW and I use MSVC/clang-cl, that will not work (due to C++ mangling).
But FMLIST should have nothing to do with a DAB-LIST. Why all this secrecy around this D/L link?

closed 2025-08-31 by KejPi