KejPi/AbracaDABra #231: DAB AF

Mike86-CZE, 2025-07-31:
Display of broadcast alternative frequencies for the selected service.

andimik, 2025-07-31:
Well, as you are a professional user, you might be able to install eti-cmdline from https://github.com/JvanKatwijk/eti-stuff (in order to get a eti from the raw signal) and https://github.com/Opendigitalradio/etisnoop

With the latter you even can filter by FIG.

andimik, 2025-09-03:
With `etisnoop -F 0/21 -i foo.eti` you should see an output like that (this is Austrian 1st national ensemble). But please note, it's transmitted quite rarely (once a minute?). Multiplexes with an SFN only do not need to transmit it.

```
FIG: 0/21
     Data:
     Length: 27
     OE: 0
     C/N: 0
     P/D: 0
     Decoding:
      -
       RegionId: 0x000
       Len: 24 Bytes
       FIs:
        -
         Length Freq list: 3
         ID field: 0xA101 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 1 , continuous output possible
         database key: 0x000659472
         Frequency Information:
          -
           195936 kHz
           geographically adjacent area
           no transmission mode signalled
        -
         Length Freq list: 6
         ID field: 0xA101 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 1 , continuous output possible
         database key: 0x000659472
         Frequency Information:
          -
           176640 kHz
           geographically adjacent area
           no transmission mode signalled
          -
           180064 kHz
           geographically adjacent area
           no transmission mode signalled
        -
         Length Freq list: 6
         ID field: 0xA101 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 1 , continuous output possible
         database key: 0x000659472
         Frequency Information:
          -
           181936 kHz
           geographically adjacent area
           no transmission mode signalled
          -
           187072 kHz
           geographically adjacent area
           no transmission mode signalled
  -
```

Or Slovene R4 (only 2 channels)

```
FIG: 0/21
     Data:
     Length: 12
     OE: 0
     C/N: 0
     P/D: 0
     Decoding:
      -
       RegionId: 0x000
       Len: 9 Bytes
       FIs:
        -
         Length Freq list: 6
         ID field: 0x9600 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 1 , continuous output possible
         database key: 0x000614400
         Frequency Information:
          -
           174928 kHz
           geographically adjacent area
           transmission mode 1
          -
           185360 kHz
           geographically adjacent area
           transmission mode 1
```

There could also be a link to other multiplexes, like on R2W Slovenia.

```
FIG: 0/21
     Data:
     Length: 17
     OE: 1
     C/N: 0
     P/D: 0
     Decoding:
      -
       RegionId: 0x000
       Len: 6 Bytes
       FIs:
        -
         Length Freq list: 3
         ID field: 0x9200 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 1 , continuous output possible
         database key: 0x4295565312
         Frequency Information:
          -
           215072 kHz
           no geographically adjacent area
           transmission mode 1

      -
       RegionId: 0x000
       Len: 6 Bytes
       FIs:
        -
         Length Freq list: 3
         ID field: 0x9300 EId
         R&M: 0x0 DAB ensemble, no local windows
         Continuity flag: 0 , continuous output not expected
         database key: 0x4295569408
         Frequency Information:
          -
           225648 kHz
           geographically adjacent area
           transmission mode 1
```

Mike86-CZE, 2025-09-03:
In the Czech Republic, regional mutations of the same multiplex are broadcast with different EIDs. Or the service is in different multiplexes. So you also need to watch FIG 0/6 and 0/24.

Mike86-CZE, 2025-10-17:
> Well, as you are a professional user, you might be able to install eti-cmdline from https://github.com/JvanKatwijk/eti-stuff (in order to get a eti from the raw signal) and https://github.com/Opendigitalradio/etisnoop
> 
> With the latter you even can filter by FIG.

Unfortunately, I can't compile the binary for Windows. I've spent a lot of time on it, but I have no experience with it.

thib66, 2025-11-28:
Hello, It would be nice to have FIG 0/21 - Service Following in the window Ensemble information here : 

[image]