KejPi/AbracaDABra #233: Feature request: Global media key support

SA-Flipperwaldt, 2025-08-08:
I would consider it convenient if it was possible to scroll through the alphabetical list of channels with the previous and next keys you find on many keyboards (and HID emulating remote controls). Preferably even if the focus (like the cursor) is somewhere else than the list. Even better if it also works when the app is in the background.

Play/pause could start/stop processing maybe. That doesn't matter as much as you can already just mute.

I'm on the Windows version 3.3.0 if that matters. Thanks for reading.

SA-Flipperwaldt, 2025-08-08:
Ah, apologies, I searched but did not come across that. I'll sleep peacefully knowing it is on the list then, thanks.

KejPi, 2025-08-08:
No need to apologize, it is not recorded here, it is in my private list 😉

closed 2026-05-10 by KejPi