KejPi/AbracaDABra #237: AppImage x64 problem since 3.1.0

FloMarl69, 2025-08-12:
Hi,

Since 11 March 2025, I have been using the 3.0.0 AppImage x86_64 release of Abracadabra, as the 3.1.0, 3.2.0 and 3.3.0 releases do not open with Zorin OS 17.3!

I noticed that the 3.0.0 AppImage x86_64 weighs 73.5 MB. Since the 3.1.0 release, it's been 67.5 MB and 68.2 MB (for 3.3.0), but it still doesn't open! Why?

andimik, 2025-08-12:
Acccording to https://en.wikipedia.org/wiki/Zorin_OS this is based on Ubuntu 22.04 LTS, which is still supported of course.

I guess it might be useful that you open a console and then start the AppImage and check if there is any error message.

Maybe `libfuse2` is missing?

```
sudo apt install libfuse2
```

KejPi, 2025-08-12:
All AppImages were built in Ubuntu 22.04 but maybe I have updated the tool I use to create the image. Have you tried to extract the image as suggested?

FloMarl69, 2025-08-13:
At this point, the situation becomes curious! Why does version 3.0.0 open and not the more recent ones?

Given that the four packages mentioned are installed on my system. Version 3.0.0 opens perfectly. I use AppImage Launcher, which extracts the image itself into its destination folder.

In the meantime, no part of ZorinOS 17.3 has been modified.

From version 3.1.0 to 3.3.0, I stupidly run the same procedure, without success.

Since version 3.1.0, there has been a decrease in size. Is there a component missing from the compilation? I extracted the 3.0.0 image without any problems using ‘--appimage-extract’, but version 3.3.0 shows this error:

`
./AbracaDABra-v3.3.0-x86_64_20ef48bd258b6809f9c80d6fa498c245.AppImage --appimage-extract
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
This doesn't look like a squashfs image.
Failed to open squashfs image
`

FloMarl69, 2025-08-13:
The fact that you told me to ‘start with a Live DVD of your distribution’ helped me find the difference. I tested it on a computer with the same distribution installed more recently.

The problem comes from a library package that was installed:
- libsquashfuse0 (0.1.103-3) : FUSE filesystem to mount squashfs archives library

Simply uninstalling this package (and its dependencies) solves the opening problem for > 3.1.0 releases.

Better still: version 3.0.0 of Abracadabra also opens. So I think that deleting the package is safe for Abracadabra (to be checked for other applications...).

I hope this is the right solution for everyone experiencing this problem.

Thank you all for your help.

KejPi, 2025-08-14:
That is very interesting resolution, thanks for sharing!
Can I close the issue?

FloMarl69, 2025-08-14:
Yes, I think yes, I can close the issue !
Thanks

closed 2025-08-14 by FloMarl69