KejPi/AbracaDABra #242: Feature request: Different tuners for different muxes

ryanginn, 2025-08-29:
Here's a bit of a daft idea I had, but I thought I'd mention it.

Would it be possible to connect 2 devices say via RTL-TCP and set them for certain frequencies? This could be useful for listening to stations from mux not close to you.

Maybe this is a bit too advanced, but here's the idea anyway:

Maybe under advanced settings:

Point RTL-TCP A to muxes 5A / 5B
Point RTL-TCP B to muxes 5C / 5D etc.

Maybe this is a weird one, but would be cool! 😎

jpuigs, 2025-08-30:
In another post, 2 days ago, I asked to have several profiles, because I have 2 RTL-TCP "servers", 1 local and 1 remote, and thanks to the developer (KejPi)  answer, I got it running.
Your idea would not work in case you have signals on same channel on both servers, like me.
I have signals on 7A on both servers....

ryanginn, 2025-08-30:
Well, in my theory you'd have assigned say 7A to one of the RTL TCP servers. Maybe a button could be added to switch between servers also.