KejPi/AbracaDABra #243: No full screen for EPG window

andimik, 2025-09-01:
The EPG windows has a maximize icon, but it cannot be set to full screen.

[image]

Tested under Ubuntu.

andimik, 2025-09-01:
strange, also the Ensemble information and Settings Window does not work.

TII, Scanner, main window work with full screen.

It's the appImage v3.3.0

KejPi, 2025-09-02:
These dialogs should not have maximize button - they do not have it neither on Windows nor on macOS, the fact that the button is displayed is somehow caused by Linux DE probably.

andimik, 2025-09-03:
Oh, more precise, it's a Ubuntu (Gnome) only problem, I just saw it correct on a different Linux laptop with Linux Mint (Cinnamon).

closed 2025-09-03 by andimik