KejPi/AbracaDABra #244: Station Logo Issue

starman316uk, 2025-09-25:
When loading a couple of stations, the station logo appears in the left hand box, but logo is so big that the program maximizes and will not let me resize and then the whole program disappears (still runs in background)
only two stations does this, other logos appear ok.

[image]

andimik, 2025-09-25:
We had this problem in Slovenia when a station used 320x320 instead of 32x32 pixels. Have asked the provider who then corrected it. It looked the same. 

[image]

KejPi, 2025-09-25:
Thanks for confirmation, I have pushed a fix that will avoid displaying oversized images that are violating standard.

StefCoders, 2025-10-22:
When is the next compile/version coming out?